iNavFlight/inav-configurator #400: A couple of things are missing from the modes tab!!!

kooopa, 2018-04-01:
Help!!

Sir. I'm having this problem too I reflashed the latest firmware v1.9.0 and redownloaded the configurator v1.9.2 I also checked if the configurator is seeing the correct version using the cli but still nothing I'm using a omnibus f3 v1.1 FC also autotune and a number of things is missing from the modes tab. what seems to be the problem? I included a screen shot of the modes tab
[image]
[image]
[image]

shellixyz, 2018-04-01:
Autotune is only available for fixed wing. Navigation modes are only available with GPS for fixed wing and with GPS+Mag for multirotors.

kooopa, 2018-04-01:
So autotune and a couple of things doesn't always show up in the modes tab it depends on what will you use it for. about the manual mode how do I find it? haha sorry I'm a noob I'm still learning hahaha anyways thanks for the reply and clarification.

shellixyz, 2018-04-01:
Yes. The manual mode is also only available when the flight controller is configured for use with a fixed wing model.

closed 2018-04-01 by kooopa