iNavFlight/inav-configurator #409: Configurator 1.9.3 issue with Windows 7 x64

digitalentity, 2018-04-21:
Reported in YouTube comments:

> Downloaded the 1.9.3 configurator and it will not open on my 64 bit win 7 laptop. Get a message saying ""Your profile can not be used because it is from a newer version of NW.js. Some features may be unavailable. Please specify a different profile directory or use a newer version of NW.js. google nw.js and it reveals its some kind of trojan virus. now none of the versions of iav config will run on this laptop. Advice please.

Lord-Quake, 2018-04-21:
I am running 1.9.3 on Windows 7 Home Premium SP 1 without any issues that I'm aware of.

MRC3742, 2018-05-02:
I have gotten this same error running Windows7 x86 (32 bit)

It happens when trying to run an earlier stand alone version of configurator which was built with an earlier version of NW.js. The error image attached below.
[image]

When you run configurator it stores the version of NW.js somewhere in the user profile. When i do a custom build of configurator it always downloads the latest version of NW.js before it proceeds to build the release. So as newer versions of configurator are released older versions will throw this error.

Fix-WorkAround?
If I delete all the User Profile files located at C:\Users\{UserName}\AppData\Local\inav-configurator
then Restart configurator and it will rebuild this folder with no more errors.

DzikuVx, 2018-07-17:
Since this is "know" solution to delete this folder, closing

closed 2018-07-17 by DzikuVx