iNavFlight/inav-configurator #412: Unable to connect for v1.9.3 on F4 board (Matek F405 STD) with iNav 1.9.1 firmware

chtay, 2018-04-24:
Prior upgrade to iNav 1.9.1, it is able to connect to F4 board with 1.9.0 firmware.
After upgrade to 1.9.1 (recent released), it failed to connect with timeout message.
Revert it back to 1.9.0, it works again.

digitalentity, 2018-04-24:
Try flashing 1.9.1 with full erase

chtay, 2018-04-24:
Done that as well...that is my default setting.

digitalentity, 2018-04-24:
Tested on my board. No issues. What do you have connected to the board?

chtay, 2018-04-24:
Hi again.
I am using Matek F405 STD. The firmware for this board is MATEKF405SE.
It is only connect my RX (Taranis XSR).
I tried to flash the firmware now (Tue 24 Apr 2018) but looks like the s3.amazonaws.com is down. I when to matek website to download but is not available as well (is there but link not available).
Possible firmware fault? Will have to wait till the link is up again.

chtay, 2018-04-24:
Hah! I saw a post in the inav github as well. My mistake in this case. In this case, lets close this issue then.

closed 2018-04-24 by chtay