iNavFlight/inav-configurator #425: OSD setting not consistent with iNav 1.9.1

Tygrys-1, 2018-05-04:
Setting total trip distance from GPS group has no effect on OSD. Also vario arrows are not shown.
They are visible in the configurator OSD preview, but not on the real OSD.

digitalentity, 2018-05-05:
@fiam @DzikuVx can you look into this?

DzikuVx, 2018-05-05:
@Tygrys-1 are you sure you uploaded latest fonts?

I just configured latest fc with latest configurator and this is working flawlesly in 2.0

Tygrys-1, 2018-05-05:
Checked once again now. Font BOLD loaded. Checked with default font also, the same result.
[image]
[image]

Tygrys-1, 2018-05-05:
I don't see the variable for a "trip distance"

Osd related dump:
# Version
# INAV/OMNIBUS 1.9.1 Apr 21 2018 / 12:53:03 (03a5c1922)

# get osd
osd_video_system = 0
Allowed range: 0 - 2

osd_row_shiftdown = 0
Allowed range: 0 - 1

osd_units = METRIC
Allowed values: IMPERIAL, METRIC, UK

osd_stats_energy_unit = MAH
Allowed values: MAH, WH

osd_rssi_alarm = 10
Allowed range: 0 - 100

osd_time_alarm = 100
Allowed range: 0 - 60

osd_alt_alarm = 1000
Allowed range: 0 - 10000

osd_artificial_horizon_reverse_roll = OFF
Allowed values: OFF, ON

osd_crosshairs_style = DEFAULT
Allowed values: DEFAULT, AIRCRAFT

osd_left_sidebar_scroll = NONE
Allowed values: NONE, ALTITUDE, GROUND_SPEED, HOME_DISTANCE

osd_right_sidebar_scroll = ALTITUDE
Allowed values: NONE, ALTITUDE, GROUND_SPEED, HOME_DISTANCE

osd_sidebar_scroll_arrows = OFF
Allowed values: OFF, ON

osd_main_voltage_pos = 258
Allowed range: 0 - 3071

osd_main_voltage_decimals = 0
Allowed range: 1 - 2

osd_rssi_pos = 83
Allowed range: 0 - 3071

osd_flytimer_pos = 311
Allowed range: 0 - 3071

osd_ontime_pos = 279
Allowed range: 0 - 3071

osd_flymode_pos = 2444
Allowed range: 0 - 3071

osd_throttle_pos = 98
Allowed range: 0 - 3071

osd_vtx_channel_pos = 386
Allowed range: 0 - 3071

osd_crosshairs_pos = 2048
Allowed range: 0 - 3071

osd_artificial_horizon_pos = 2248
Allowed range: 0 - 3071

osd_horizon_sidebars_pos = 2248
Allowed range: 0 - 3071

osd_current_draw_pos = 97
Allowed range: 0 - 3071

osd_mah_drawn_pos = 129
Allowed range: 0 - 3071

osd_wh_drawn_pos = 161
Allowed range: 0 - 3071

osd_bat_remaining_capacity_pos = 193
Allowed range: 0 - 3071

osd_bat_remaining_percent_pos = 323
Allowed range: 0 - 3071

osd_craft_name_pos = 41
Allowed range: 0 - 3071

osd_gps_speed_pos = 2243
Allowed range: 0 - 3071

osd_gps_sats_pos = 2135
Allowed range: 0 - 3071

osd_gps_lon_pos = 2065
Allowed range: 0 - 3071

osd_gps_lat_pos = 2050
Allowed range: 0 - 3071

osd_home_dir_pos = 2382
Allowed range: 0 - 3071

osd_home_dist_pos = 2384
Allowed range: 0 - 3071

osd_altitude_pos = 2199
Allowed range: 0 - 3071

osd_vario_pos = 183
Allowed range: 0 - 3071

osd_vario_num_pos = 2263
Allowed range: 0 - 3071

osd_pid_roll_pos = 34
Allowed range: 0 - 3071

osd_pid_pitch_pos = 47
Allowed range: 0 - 3071

osd_pid_yaw_pos = 105
Allowed range: 0 - 3071

osd_power_pos = 47
Allowed range: 0 - 3071

osd_air_speed_pos = 163
Allowed range: 0 - 3071

osd_ontime_flytime_pos = 2391
Allowed range: 0 - 3071

osd_rtc_time_pos = 407
Allowed range: 0 - 3071

osd_messages_pos = 2401
Allowed range: 0 - 3071

osd_gps_hdop_pos = 119
Allowed range: 0 - 3071

osd_main_cell_voltage_pos = 290
Allowed range: 0 - 3071

osd_throttle_auto_thr_pos = 2178
Allowed range: 0 - 3071

osd_heading_graph_pos = 107
Allowed range: 0 - 3071

osd_efficiency_mah_pos = 161
Allowed range: 0 - 3071

osd_efficiency_wh_pos = 161
Allowed range: 0 - 3071

digitalentity, 2018-05-05:
Ok, not a configurator bug. Trip distance is not shown when current sensor is disabled.

closed 2018-05-05 by digitalentity

digitalentity, 2018-05-05:
For reference: https://github.com/iNavFlight/inav/issues/3162, https://github.com/iNavFlight/inav/pull/3163

Tygrys-1, 2018-05-05:
Confirmed for vario and distance. Thank you for the solution. Issue solved/explained.