iNavFlight/inav-configurator #430: commit 343e4d244d498bfafe6ad32526764be4b9a9e429 is breaking connection on linux

eephyne, 2018-05-06:
starting at this commit, my configurator can't connect to the fc, before its working just fine.

when its not connecting its stay on waiting for data.

I can provide more info if needed.

I'm using Arch linux with node 9.8 (using nvm)

shellixyz, 2018-05-06:
Working fine for me. You need firmware version <= 1.9.1or at least 49b34440 in the dev branch for it to work. If you have a version in between it is normal it hangs.

eephyne, 2018-05-07:
I do have version 2.0 (thats the reason I use git version of configurator)

digitalentity, 2018-05-07:
Can't reproduce the issue. Latest Configurator + latest development firmware is working for me.

eephyne, 2018-05-07:
How can I debug this to see where it come from?

DzikuVx, 2018-05-07:
I have exactly the same situation over here. The error is:

```
Error in event handler for serial.onReceive: RangeError: Offset is outside the bounds of the DataView
    at Object.self.processData (chrome-extension://pdhhfdfdagihaglfkggflmpdpnaejabp/build/script.js:9384:55)
    at Object._dispatch_message (chrome-extension://pdhhfdfdagihaglfkggflmpdpnaejabp/build/script.js:15278:23)
    at Object.read (chrome-extension://pdhhfdfdagihaglfkggflmpdpnaejabp/build/script.js:15257:26)
    at read_serial (chrome-extension://pdhhfdfdagihaglfkggflmpdpnaejabp/build/script.js:13784:13)
```

DzikuVx, 2018-05-07:
@shellixyz @eephyne  apparently this is happening when FC is not latest development = does not have latest changed for a battery.

I wonder if we can fail more gracefully in such a cases.

@eephyne I'm gonna close it since it is just happening due to mismatch and latest development of FC fixes the issue

closed 2018-05-07 by DzikuVx