iNavFlight/inav-configurator #445: iNav 1.9.3 Sensor tab showing wrong values

Richrfl, 2018-05-12:
In that screen, accelerometer graphics shows gyro values and vice versa.

Don't know if this is the right place for suggestions, so, apologizes in advance.
Suggestions: 
It would be nice to be able to change configurations, without the need of being connected to the board.
I guess a rudimentary way of saving a configuration is to copy a dump into a txt file, and it should be nice to bring it back to the configurator to make changes on it, and later to upload it to the board.

DzikuVx, 2018-05-13:
[image]

Just tested FC 1.9.1 and Configurator 1.9.3, Sensor tab works like expected

closed 2018-05-13 by DzikuVx

Richrfl, 2018-05-13:
Pawel, I don't know if I am totally confused, but gyroscope values should
show attitude, meaning, if the board is rolled say, 3- degrees, the gyro
should follow the board attitude, while the accelerometer should only show
a "jump" only when the board changes attitude, but is the board stays in a
30 degree angle, the accelerometer will show zero value (or around that,
given the sensibility).
In other terms, the gyro is supposed to control the "artificial horizon",
or I learnt all upside down?
Anyway, the board makes the right corrections both for unexpected changes
of attitude (like wind gusts) or auto level.
Thank you for your response.

DzikuVx, 2018-05-13:
No, gyroscope show rotation speed. If there is not rotation movement, gyro reads ZERO. Acc reads current acceleration. So when lying flat, Z axis should read 1G.

Richrfl, 2018-05-14:
Pawel, I regret to keep this exchange because I feel I am using your time,
but this discussion is educating, and I think there is a lexicon
difference. According with what I learnt, the accelerometer operation is
based in force applied over some crystals, that produce an electrical
signal. Now, Newton tells me the there is a force (or acceleration,
positive or negative) when a movil changes direction or speed in a linear
movement. With that said, if I am flying level and horizontal, at constant
speed, there is no acceleration. Now, I make a 90 degrees roll, and enter
into a knife edge. While I am rolling I will have an acceleration but while
a stay in the knife, there is no acceleration. On the other hand in a
mechanical gyroscope, the turning wheel stays in a given position,
regardless how the rings are moved, and the difference between the rotating
wheel and the new position of the rings, give the attitude. In the case of
the knife, there will be a constant 90 degrees difference between the
original position and the final position, meaning that I will have a
deviation signal for as long as I keep flying straight in the knife.
However, here I have to declare my ignorance on how an electronic gyroscope
works, but may be you can give me some additional knowledge on how real
things work.
Please, receive my gratitude for your time, and my apologies for entertain
you on this, but I am a constant learning person. They call it scientific
mind set.

DzikuVx, 2018-05-14:
MEMS gyros are 6 axis devices. Gyro part measures rotation speed aka angular velocity i X,Y and Z axises relative to the gyro itself. In degrees per second. No angular velocity, no gyro output.

Accelerometer part measures acceleration in X, Y and Z (again relative to device itself) including earth acceleration. Acc will read zero in all axises only in free fall. 

So, everything is correct.

Richrfl, 2018-05-14:
Thanks a lot Pawel. This was educating.
Have a good one.