iNavFlight/inav-configurator #455: Suggestion: Add short help (?) for more options.

Tygrys-1, 2018-05-22:
Some options does not have the question mark help, and not always the option name is self-explaining it's purpose.

Tygrys-1, 2018-05-23:
:-) There is a kind of paradox. If I will know and understand things well, I'll not need the context help. And when I need help to understand things, I am not able to explain them to others yet. My level of understanding is still on the "need to know" basis. 

I think that some things are well explained in the wiki in the variables section, so my first guess is just to move them to the (?) marks bubles in the configurator.

DzikuVx, 2018-05-23:
I'm in the same position for example. I do not know what I do not know,
right?

closed 2019-03-24 by shellixyz