iNavFlight/inav-configurator #468: Page not displaying correctly on notebook

flyfisher2, 2018-06-08:
The linux version of Configurator 2.0 cuts off the bottom of the page when run on a notebook with a display resolution of 1366 x 768 (the best my HP notepad can do).
The build was kindly provided by @stronnag, from git commit f31db904.

The WIN32 version displays the whole page correctly.

flyfisher2, 2018-07-01:
Any chance someone could look at this?
I have all my iNav related apps (except the configurator) running on a linux notebook and would like to be able to use the configurator to make changes at the flying field.

powercycle, 2018-12-02:
Flyfisher2,

Try using xrandr

xrandr --fb 1600x1024 --output LVDS1 --mode 1366x768 --panning 1600x1024

Worked for me until they are able to find a solution.

xrandr by itself will indicate which settings you might need to change for your setup.

flyfisher2, 2018-12-02:
Thanks for that.
I haven't been using that old notebook recently but I will dig it out and give this a try.

DzikuVx, 2020-03-03:
#942 min windows size lowered to 800x600 now

closed 2020-03-03 by DzikuVx