iNavFlight/inav-configurator #470: Change i2c1 on SpracingF3 Evo to i2c2 using Omibus i2c uart 3 settings possible?

markpmasonst, 2018-06-09:
I'm wondering if I can use the PB10 PB11 pins on uart3 to use on spracingf3 evo for i2c2 if I compile using the i2c code from omnibus or other targets that use it as an option? I'm having problems compiling but that's another issue, but I'm not going to bother anyone with that right now so I thought I'd ask about this. I know there are a lot of people that would like i2c on brushed boards that are not broken out. This seems like it would be an easy custom solution. The board I'm using currently uses these pins on uart 3.

shellixyz, 2018-06-13:
You cannot use PB10/PB11 for I²C. I²C on the OMNIBUS board is on PB6/PB7. Hardware I²C is only available on specific pins.

markpmasonst, 2018-06-13:
Hi Michel,

Yes, I was mistaken. It wasn't the Ominbus. The Ominbus F4 uses I2C2 on
uart3 using PB10 and 11. If you check the .OmnibusF4 Target.h it clearly
shows this. My compiler is working now so I'll check it later tonight.
Sorry about the confusion.

Mark Peeling

DzikuVx, 2018-06-14:
I'm closing this since it is a firmware not Configurator. Thanks

closed 2018-06-14 by DzikuVx