iNavFlight/inav-configurator #471: Are Acc and Gyro x and y sensor display labels reversed?

KiwiDavid, 2018-06-12:
Playing with a bare SPF3 board and referring to the manufacturers axis data (MPU-6050) it appears that the configurator may have acc and gyro x and y axis reversed.

The change between z axis facing up and z axis facing down should be and is 2G.   With the chip's x axis facing up and then facing down the difference on the sensor's x display is zero however the y axis changes through 2G.
Similarly when doing a quick rotation about the z axis the z gyro responds whereas when doing a quick rotation about the chip's x axis it's the y gyro plot that responds.
It may be that this is academic due to the calibration process and it certainly does not affect performance however it may make any diagnostic process a little confusing.
Can someone take a look and see if they can confirm what I'm seeing?

David

DzikuVx, 2018-06-14:
`MSP_RAW_IMU` reports in board, not chip orientation, so this is probably the reason for what you are observing

closed 2018-06-14 by DzikuVx

digitalentity, 2018-06-14:
> Thanks for the reply.
> OK I see where you're coming from. So X axis is Pitch and Y axis is Roll.
> So when the plane is swung between max and min pitch it's actually max and
> min Y output of the chip. Perhaps the graph should have pitch, roll, yaw?
>
> Best regards,
> David

Pitching will be read as rotation around Y axis and gravity change on X
axis.

Let's not mix up sensors, there's no roll/pitch/yaw for accelerometer and
compass - only X/Y/Z axis.

KiwiDavid, 2018-06-14:
Can you please confirm -  that's Y axis and X axis as displayed, not actual chip axis outputs?

digitalentity, 2018-06-15:
@KiwiDavid what would be the point in showing sensor axis? Firmware is only concerned about aircraft-relative data. Data you see are even board-relative because they consider board orientation as well.

KiwiDavid, 2018-06-15:
OK thanks for clearing that up.  Aircraft-relative are the key words.   Sorry to be a pain on this.

David