iNavFlight/inav-configurator #501: Configurator 2.0.0.RC1- RC channels missing from the Mixer

Arvycka, 2018-06-28:
2.0.0-RC1
SBUS 16 ch. RX
Only RC channels 5 to 8 are available in the mixer.
Trims are setup on channels 8 and 9 for "live" trimming in previous Configurator versions.
How to forward these channels to servos in the new mixer?

closed 2018-07-16 by Arvycka

blurcak, 2018-08-01:
Hi why U closed this issue? the servo mix is still limited to channels 1-8 on TX, im using ch10 for servo tilt, and its not accesible

Arvycka, 2018-08-01:
I  decided that I can remap channels and use 5, 6, 7 and 8 for auxiliary functions. I did not expect this issue to be fixed.

blurcak, 2018-08-01:
Ohh, ok thats solution too :), im missing old mixer and channel forward :)

Arvycka, 2018-08-01:
New channel forwarding works just fine, only devs forgot to include the rest of the channels. ☺ Or maybe there are technical reasons for it. Either way this issue did not get any attention...