iNavFlight/inav-configurator #505: Motors tab not working in 2.0 rc1

mirohristov-com, 2018-06-28:
The motors tab doesn't show individual motors and the master slider doesn't spin the motors. 
(with the safety checkbox on)

FC: Omnibus F4 v2 Pro.

Should I try iNav 1.9?

[image]

flyfisher2, 2018-06-29:
I had that once when I first upgraded to 2.0.
Just powered everything OFF and On, been OK ever since.

DzikuVx, 2018-06-29:
@mirohristov-com you are not using INAV 2.0. Only INAV 2.0 with Configurator 2.0
`Running firmware released at Apr 21...`

closed 2018-06-29 by DzikuVx

mirohristov-com, 2018-06-29:
Omg i'm an idiot :) I just re-flashed and it works like a charm.

Quick question: is there a reason the unstable releases are listed as boards with a '-rc1' prefix as opposed to firmware version under that board?

[image]

DzikuVx, 2018-06-29:
This is bug that will be solved in RC2