iNavFlight/inav-configurator #512: 2.0.0 RC2 inav-configurator.exe can not start

davidngrc, 2018-06-30:
I try use the  INAV-Configurator_win32_2.0.0-rc2.zip
but it just show the nw.js start up screen.
I then found out this folder
C:\Users\David\AppData\Local\inav-configurator

I rename it to 
C:\Users\David\AppData\Local\inav-configurator_bak

then run the rc2 inav-configurator.exe again, it work this time.
so maybe in the future startup, check if the config folder exists, if yes, ask user if he/she want to delete it or not?   or rename the old config folder to inav-configurator_backup_date_time ？

mickey2011, 2018-06-30:
Doesn't matter what I do I can't run RC2 configurator, it is always minimised

davidngrc, 2018-06-30:
have you delete the inav-configurator folder?
if you can not find it, search your c drive with that folder name, delete anything you find.

DzikuVx, 2018-06-30:
https://github.com/iNavFlight/inav-configurator/blob/master/README.md. 

Second paragraph

closed 2018-06-30 by DzikuVx