iNavFlight/inav-configurator #514: new mixer does not display tricopter motor allocation correctly (nor reversed servo)

stronnag, 2018-07-01:
"One of our motors is missing"

[image]

```
# INAV/MATEKF405 2.0.0 Jul  1 2018 / 09:19:28 (8f16ac1ff)

mmix 0  1.000  0.000  1.333  0.000
mmix 1  1.000 -1.000 -0.667  0.000
mmix 2  1.000  1.000 -0.667  0.000

# servo mix
smix 0 5 2 -100 0
set platform_type = TRICOPTER
set model_preview_type = 1
```
The example shown flies perfectly.

However, this appears to be causing some concern to the CLI challenged on RCG.

Noted that the reversed servo is also not shown as reversed in the servo tab, yet the weight /rate is +ve there (can we also be consistent in our nomenclature, it also causes some users much confusion).

[image]

stronnag, 2018-12-27:
The reversed servo display is fixed (now shows -100) at least in 08d50f7feec3631592e30cc45e92788966c8a15b

DzikuVx, 2018-12-28:
thanks @stronnag for testing

closed 2018-12-28 by DzikuVx

stronnag, 2018-12-28:
A little misunderstanding. The above comment only showed that half the problem is addressed (the reversed servo).

The PWM allocation is still displayed incorrectly (shows Servo5, Motor 2, Motor3) --- motor 1 is absent. 

[image]