iNavFlight/inav-configurator #516: Feature Request: Format blackbox SD Card and export logs

davidngrc, 2018-07-01:
It would be nice if iNav configurator have the following features
1. Format blackbox SD Card
2. Export blackbox SD Card logs

Afinogen, 2018-07-12:
I support, the SD card is not always convenient to take out

fiam, 2018-07-12:
I did some work in this INAV branch, but I got file corruption problems. If anyone wants to take a crack at it, be my guest: https://github.com/iNavFlight/inav/tree/agh_msc

davidngrc, 2018-07-15:
new to this, too many commits, could you point me to the right one?

davidngrc, 2018-07-22:
@fiam ?

Afinogen, 2018-07-24:
Get list logs from SD to configurator - success.
Next - read log file...

davidngrc, 2018-07-24:
@Afinogen nice one, so you modify the iNav code or just modify the inav-configurator code?

Afinogen, 2018-07-24:
@davidngrc  and iNav and inav configrurator code. But experience in the development for microprocessors I do not, I will need code review)
After work I push the branches for viewing

Afinogen, 2018-07-24:
@davidngrc  https://www.youtube.com/watch?v=LMbChvbcZzw
Is there any reason?

davidngrc, 2018-08-06:
@Afinogen that method is better, but is it hard to implement here?

Afinogen, 2018-08-07:
@davidngrc I could not implement msc ( The option with msc seems to me to be the best.
I managed to get the list of files in the CLI (using asyncfatfs), but there are problems when working with msp. I can not solve this problem yet and there is not enough time

Jonny360, 2019-12-08:
Yes plz, after spending 10 minutes with tweezers to get an sdcard in only to realize I can't access from usb. Please implement this.

AdamGFPV, 2020-08-25:
Any news on this? It would be so useful as I cannot get to the SD card on all of my builds

MrD-RC, 2022-11-26:
MSC is implemented in 5.0

closed 2022-11-26 by MrD-RC