iNavFlight/inav-configurator #518: version missing n connect , cannot enter Mixer

stefanolofsson, 2018-07-03:
2018-07-03 @ 09:45:42 -- Serial port successfully opened with ID: 1
2018-07-03 @ 09:45:42 -- MultiWii API version received - 2.1.0
2018-07-03 @ 09:45:42 -- Flight controller info, identifier: INAV, version:
2018-07-03 @ 09:45:42 -- Running firmware released on: Jul 2 2018 21:19:17
2018-07-03 @ 09:45:42 -- Board: OB43, version: 0
2018-07-03 @ 09:45:42 -- Unique device ID received - 0x2200463436510b38363236

Tygrys-1, 2018-07-03:
Are you trying to use the configurator v2.0 with inav 1.9.1?

stefanolofsson, 2018-07-03:
Yes, i was. I am on RC2 now. I can enter mixer page but still no report of version.

2018-07-03 @ 13:21:57 -- MultiWii API version received - 2.2.0
2018-07-03 @ 13:21:57 -- Flight controller info, identifier: INAV, version:
2018-07-03 @ 13:21:57 -- Running firmware released on: Jun 29 2018 12:35:56
2018-07-03 @ 13:21:57 -- Board: OB43, version: 0

I guess this can be closed. Thanks for you reply

closed 2018-07-03 by stefanolofsson

Tygrys-1, 2018-07-03:
From time to time I have the same problem with inav 1.9.1 and configurator 1.9.3. Sometimes even worse - I can run only CLI or the configurator say, that I must have Inav installed. Disconnecting+connecting the board solves the problem. Must be some timing issue or maybe a garbage in the buffer.