iNavFlight/inav-configurator #522: 2.0.0-rc2 Packet Errors

teckel12, 2018-07-04:
Running dev INAV 2.0.0 RC2 firmware and getting a ton of packet errors in the configurator.  Stuff just never works or takes a very long time to work.  Didn't have a problem with RC1.  I heard one other person reporting the same issue but can find no mention of it here.

fiam, 2018-07-04:
Either configurator RC1 and inav RC2 or viceversa. Please, use the same RC for both things, since we can’t do version checks based on RCs for now, there’s no way to make the configurator compatible with both (they both report 2.0.0)

teckel12, 2018-07-04:
It was my own build from development INAV and release configurator RC2.  But, who knows exactly where my development build was from (I know I re-synced after RC2).  I just did another re-sync with the development branch, flashed and now there's no more errors.  So, I probably had a mix-mash of RC1/RC2 in my development branch.

I also didn't think that this is how a version mix-mash would manifest.  It seemed there was some other core problem.  Also, interesting that doing a closed issue search for `packet` yielded no relevant results.  Seems this would be something that someone reported at some point?

closed 2018-07-04 by teckel12