iNavFlight/inav-configurator #525: Request: Add mini talon, 1.2 and 2 meter flying wing preset.

davidngrc, 2018-07-08:
mini talon is a popular plane.
and the Sonicmodell HD Wing, and Skywalker X8
Thank you.

DzikuVx, 2018-07-08:
If you have confirmed configuration for those airplanes @davidngrc we can add them. If not, then well, maybe someone else has? If not, then well, I would prefer not to guess

davidngrc, 2018-07-08:
I dont have it.
Lets hope someone will add it here.

Afinogen, 2018-07-12:
I think enough servomix. The rest depends on the equipment

wx4cb, 2018-07-22:
i've found that with mine, i had to change the rates for the ailerons as no matter what I did they would turn the wrong way. Here's my SMIX. By default i think everything was -100

# servo mix
smix 0 2 0 100 0
smix 1 3 0 100 0
smix 2 4 1 100 0
smix 3 5 1 -100 0
smix 4 4 2 -100 0
smix 5 5 2 -100 0

wx4cb, 2018-07-22:
Here is my "diff", if y'all see anything i missed or should change that would be great.

# diff

# version
# INAV/MATEKF405SE 2.0.0 Jul 14 2018 / 22:38:02 (f2cf315ad)

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 2 0 100 0
smix 1 3 0 100 0
smix 2 4 1 100 0
smix 3 5 1 -100 0
smix 4 4 2 -100 0
smix 5 5 2 -100 0

# servo
servo 2 1000 2000 1500 50
servo 3 1000 2000 1500 50

# feature
feature -SOFTSERIAL
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 0 0 57600 38400 0 115200
serial 1 2 115200 57600 0 115200
serial 2 256 115200 38400 57600 115200
serial 4 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1850 2100
aux 1 2 1 1300 1700
aux 2 3 2 1300 1700
aux 3 3 1 1800 2100
aux 4 9 2 1300 1700
aux 5 8 2 1825 2100
aux 6 35 1 1800 2100
aux 7 10 1 900 1175
aux 8 27 3 1800 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 24 0 V
osd_layout 0 1 1 0 V
osd_layout 0 4 8 6 H
osd_layout 0 6 7 14 V
osd_layout 0 7 1 13 V
osd_layout 0 9 1 2 H
osd_layout 0 11 8 0 V
osd_layout 0 12 1 1 V
osd_layout 0 13 1 6 V
osd_layout 0 14 15 13 V
osd_layout 0 15 1 5 V
osd_layout 0 20 21 14 V
osd_layout 0 21 21 13 V
osd_layout 0 22 14 0 V
osd_layout 0 24 16 0 V
osd_layout 0 26 1 7 V
osd_layout 0 28 23 11 H
osd_layout 0 29 7 13 V
osd_layout 0 30 0 10 V
osd_layout 0 31 15 14 V
osd_layout 0 33 14 1 V
osd_layout 0 35 7 1 V
osd_layout 0 40 22 1 V
osd_layout 0 44 0 0 V

# master
set acc_hardware = MPU6000
set acczero_x = 23
set acczero_y = -7
set acczero_z = -337
set accgain_x = 4081
set accgain_y = 4088
set accgain_z = 4042
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set failsafe_procedure = RTH
set model_preview_type = 28
set gps_sbas_mode = AUTO
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 9144
set nav_fw_cruise_thr = 1600
set nav_fw_max_thr = 1900
set nav_fw_bank_angle = 30
set osd_units = IMPERIAL
set osd_crosshairs_style = AIRCRAFT
set osd_left_sidebar_scroll = GROUND_SPEED
set osd_right_sidebar_scroll = ALTITUDE
set osd_main_voltage_decimals = 2
set osd_coordinate_digits = 8

# profile
profile 1

set roll_rate = 60

# battery_profile
battery_profile 1

set battery_capacity = 10000
set battery_capacity_warning = 3000
set battery_capacity_critical = 2000

kardon18, 2018-12-10:
> 
> 
> If you have confirmed configuration for those airplanes @davidngrc we can add them. If not, then well, maybe someone else has? If not, then well, I would prefer not to guess

Regarding the ZOHD Nano Talon
I would recommend that **RC Yaw** replace **Yaw Stabilization**, under the mixer presets.

This little plane is NOT at all suited to Yaw axis's stabilization. 
I have found that by disabling it, and just using **RC yaw** on the rudder mix. Makes a large difference to the way this plane holds its heading in crosswind conditions. 
Plus **RC Yaw** will still provide manual rudder control.

I have also tuned the Pitch axis PIFF. More like a PID. Which also help this particular plane considerably. Due to the Nano Talon being very sensitive to Pitch axis "P" term gain increases. This in turn cause issues if the FF is cranked up even the slightest. Which prevents pitch PIFF working to its optimum on this plane.

[image]
[image]

closed 2019-05-24 by DzikuVx