iNavFlight/inav-configurator #531: `gulp dist` fails

digitalentity, 2018-07-09:
```
$ git show
commit f48c6a0802da022fc8fc6b9bd8a03203bdbd9970 (HEAD -> master, origin/master, origin/HEAD)
Merge: b8fd9578 39beafcb
Author: Alberto García Hierro <alberto@garciahierro.com>
Date:   Mon Jul 9 00:55:02 2018 +0100

    Merge pull request #523 from OlivierC-FR/font-vision

    OSD font : Vision
```

```
$ gulp dist
[20:13:38] Using gulpfile D:\Dev\CYGWIN\home\bitph\inav-configurator\gulpfile.js
[20:13:38] Starting 'dist'...
[20:13:38] 'dist' errored after 3.35 ms
[20:13:38] TypeError: gulp.hasTask is not a function
    at D:\Dev\CYGWIN\home\bitph\inav-configurator\node_modules\run-sequence\index.js:23:22
    at Array.forEach (<anonymous>)
    at verifyTaskSets (D:\Dev\CYGWIN\home\bitph\inav-configurator\node_modules\run-sequence\index.js:17:11)
    at runSequence (D:\Dev\CYGWIN\home\bitph\inav-configurator\node_modules\run-sequence\index.js:130:2)
    at D:\Dev\CYGWIN\home\bitph\inav-configurator\gulpfile.js:217:12
    at taskWrapper (D:\Dev\CYGWIN\home\bitph\inav-configurator\node_modules\undertaker\lib\set-task.js:13:15)
    at bound (domain.js:301:14)
    at runBound (domain.js:314:12)
    at asyncRunner (D:\Dev\CYGWIN\home\bitph\inav-configurator\node_modules\async-done\index.js:55:18)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
```

digitalentity, 2018-07-09:
Might be related https://stackoverflow.com/questions/48513573/gulp-error-gulp-hastask-is-not-a-function

`npm install` installes gulp-4.0.0

fiam, 2018-07-09:
I’ll take a look later. I upgraded some parts of the gulpfile to gulp 4 but didn’t test all tasks.

commit 321d7046 (2018-07-09): Complete migration to Gulp 4, add linux32 support

- Remove usages of runSequence(), which doesn't work with Gulp 4,
using gulp.series() instead.
- Add support for linux32 builds
- Add new --platform argument for building only for a single platform
(e.g. --platform=osx64).

Fixes build problems introduced in #527
Fixes #531
PR #532 (2018-07-09): Complete migration to Gulp 4, add linux32 support
- Remove usages of runSequence(), which doesn't work with Gulp 4,
using gulp.series() instead.
- Add support for linux32 builds
- Add new --platform argument for building only for a single platform
(e.g. --platform=osx64).

Fixes build problems introduced in #527
Fixes #531
closed 2018-07-09 by fiam