iNavFlight/inav-configurator #545: Feature Request: UI indicator on RSSI, SBUS, SmartPort wire is connected to the receiver (if that feature is enabled)

davidngrc, 2018-07-16:
Recently, I had problem with telemetry, I had no idea why it does not work.
turn out it was the telemetry wire got loose.

Could we have an indicate like the "top GPS icon" in the menu?
I am sure it is easy to implement, like gps, frsky receiver sbus, smartport, rssi wire will send data to the flight controller every second.
We just need a listener, count those messages, if got any success message, then green icon, otherwise, red icon.

davidngrc, 2018-07-17:
Ref to this loose wire issue https://github.com/iNavFlight/inav/issues/3565

DzikuVx, 2019-05-24:
I'm affraid this is a no-go. Telemetry is usually FC->radio  not the other way around and FC might just not know that Radio is not reading anything. Sure, S.Port is bi directional but that does not mean that FC knows that radio received a signal. RSSI on the other had is either is either injected in to one channel or via ADC. ADC has no idea that wire is connected or not, receiver is powered up or sending any data. Configurator already show read RSSI in % but interpretation is up to user

closed 2019-05-24 by DzikuVx