iNavFlight/inav-configurator #548: iNav Configurator does not open on Windows

bs1957, 2018-07-20:
Version 2.0.0-rc3 Win 32
OS: Windows 7 Ultimate
RAM 4GB

Program starts and closes its window immediately. Icon on the task bar stays on.
Task manager shows 4 exe instances running though I started it once.
debug log: 
[0719/210755.226:ERROR:registration_protocol_win.cc(56)] CreateFile: The system cannot find the file specified. (0x2)
[0719/210800.772:ERROR:registration_protocol_win.cc(56)] CreateFile: The system cannot find the file specified. (0x2)

DzikuVx, 2018-07-20:
https://youtu.be/XMoULyiFDp4

closed 2018-07-20 by DzikuVx

killa198921, 2020-05-18:
I’m having the same issue.

DzikuVx, 2020-05-19:
@killa198921 the last comment shows the solution to the problem

Krilatii, 2021-01-14:
Win 7 -64. Такая же проблема! второй день разбираюсь, не могу запустить!

Krilatii, 2021-01-14:
Спасибо ну тут я не нашел решения проблемы. человек на видео решает ее в win 10.. у меня нет таких файлов и папок

NigelDcruz, 2023-02-11:
> https://youtu.be/XMoULyiFDp4

This solved the problem. Thanks.