iNavFlight/inav-configurator #555: [Feature Request] Modes ordered if used

wx4cb, 2018-07-22:
Not sure how easy this would be to implement, but on the modes screen, would it be possible to have the modes that you are using to be ordered to the top of the list at page load?

for example, see the attached picture, it would be nice that the modes that are being used were all at the top to avoid scrolling half/3-4' of the way down the page. It would help with knowing which ones were on a particular switch and the logic etc - at least for me.

It doesn't have to be "dynamic", but could be done whenever the page loads and gets the new "values"

[image]

Hennyr, 2018-07-22:
I agree, for mee to

Tygrys-1, 2018-07-31:
I'd like a chcckbox for "Sort the used modes first". When assigning modes it is good to have them all visible in the known order. When managing used modes it might be convinient to have used on top.

closed 2021-06-26 by DzikuVx