iNavFlight/inav-configurator #565: Feature Request: in Modes tab, highlight the mode in yellow color if the pwm signal is received but not active.

davidngrc, 2018-08-17:
if the mode is active, now it will highlight the relate mode in blue.

some other mode like "NAV ALTHOLD", "NAV POSHOLD", "NAV RTH".... will not highlight in blue due to some reason, like not get a gps lock.

how about we highlight those mode in yellow if the pwm signal is received?
so that we know the switch works by just looking at the color changing?
(yellow usually means not ready or soon to be ready, like the traffic light)

rsilk1949, 2020-02-25:
Suggest changing colours from blue to green for active and yellow for selected but not active.

MrD-RC, 2020-10-29:
This is something that is really needed. The modes page really confuses new people. For example, the put the switch in manual and manual lights up blue. The put the switch in RTH but is stay grey. Because of this, they think that RTH won't work. They don't realise that it is not changing colour because the flight controller can't enter the mode.

Having feedback to show the switch positions should work when all the conditions are met in the flight controller, I think, is vital.

closed 2022-03-14 by DzikuVx

MrD-RC, 2022-03-14:
Dark blue is in range. Light blue is active on the FC. Been there since 3.0 or 4.0. Can’t remember which.