iNavFlight/inav-configurator #567: feature request: copy osd screens

P-I-Engineer, 2018-08-23:
it would be cool to be able to copy the osd layout between the screen layouts.

it's an easy way to make pop up osd items.

thank you

wx4cb, 2018-11-18:
having just setup my osd screens, i agree, it would be nice to be able to copy into the others. I have some things such as battery voltage and lat/long on all the screens int he same place.

shellixyz, 2018-11-18:
For now you can do it from the CLI. You can find the ids here: https://github.com/iNavFlight/inav/blob/master/src/main/io/osd.h#L38

For example if you want to have VBAT (id 1) at the same position on all the OSD layouts you need to run:
osd_layout 0 1 column row V
osd_layout 1 1 column row V
osd_layout 2 1 column row V
osd_layout 3 1 column row V

jj-uk, 2019-05-07:
+1 on this. I have setup the default layout the way I like it. I'd now like to duplicate that to 2 alt osd screens, then turn off stuff on them.

A "copy from OSD x screen" button next to the OSD preview would be a great addition.

pDR8iLQ4go, 2020-03-14:
Does it work to paste a part of a dump file like „osd_layout 0 1 column row V“ in the CLI command line? Or may I even copy and paste all osd information in the CLI command line?

jj-uk, 2020-03-14:
Yes. Dump to a text file, and change the screen I'd then paste back in the terminal.

MrD-RC, 2022-11-26:
https://www.mrd-rc.com/tutorials-tools-and-testing/useful-tools/inav-osd-switcher-tool/

closed 2022-11-26 by MrD-RC

jj-uk, 2022-12-29:
I don't think this should have been closed. It's not been implemented in iNav. Creating an external tool is a cool, but having this in the GUI is cooler.