iNavFlight/inav-configurator #568: inav config for chrome 2.0.0_0 don't  work in ubuntu 16.04

audiserg, 2018-08-25:
please add to chrome application repository old version (1.9.3), new version start but can't connect.

Hennyr, 2018-08-28:
I am using version 2.0 under Ubuntu 18.04 64bits, thru Chrome-browser and is working fine.

pierpiero83, 2018-12-02:
I downloaded the configurator for my system Ubuntu 18.04, extracted the folder, put everything in /opt but when I click on "inav-configurator" it does not run the application. I am new to the linux world and I would like to run the program on it.
Any help?

shellixyz, 2018-12-02:
@pierpiero83 For some reason it looks like the executable is recognized as a shared library so that's why it doesn't run. You should be able to run it from a terminal.

- Open a terminal
- type `cd installation_directory` replacing `installation_directory` with the directory you installed the configurator in, to switch to the configurator directory
- type `chmod +x inav-configurator` (you only need to do this once)
- then run the configurator by typing `./inav-configurator`

pierpiero83, 2018-12-02:
> @pierpiero83 For some reason it looks like the executable is recognized as a shared library so that's why it doesn't run. You should be able to run it from a terminal.
>
> * Open a terminal
> * type `cd [...]

Thank you... it works now...

closed 2019-03-24 by shellixyz