iNavFlight/inav-configurator #569: configurator doesn't work in 10.9.5 HELP !!!

freddybloggs, 2018-08-27:
I have 10.9.5 you have disabled the standalone app on my machine.

shellixyz, 2018-08-30:
I'm using 50.29.32.45 and it is working fine for me :rofl:

freddybloggs, 2018-08-30:
what is 50.29.32.45 and where do I get it, the last usable version for me is 1.9.3 and with this I am unable to set up my HEX as there is no mixer TAB.

shellixyz, 2018-08-30:
You can get the latest version here: https://github.com/iNavFlight/inav-configurator/releases

freddybloggs, 2018-08-30:
I have it, it won't work with 10.9.5 is says it requires OS 10.10 or greater

shellixyz, 2018-08-30:
Latest is 2.0.0. You are sure you got the latest one ? Screenshot ?

freddybloggs, 2018-08-30:
[image]

shellixyz, 2018-08-30:
Maybe it is a requirement of the updated NPM

closed 2019-03-24 by shellixyz