iNavFlight/inav-configurator #570: Add some additional GUI options for Fixed Wing

b14ckyy, 2018-09-06:
Although iNAV has nice main profile type selection, some important changes are not accessible via GUI or optimized for fixed wing

Examples: 

- nav_wp_radius > should be higher for fixed wing per Default and be configurable in GUI in General
- nav_wp_safe_distance > could also be higher for fixed wing like 200m instead of 100m bwcause the plane could Need some more way to go to some Navigation altitude
- nav_fw_cruise_speed > why is that not in GUI already? Would be more useful than cruise throttle because if some stronger wind appears, cruise throttle could be not enough. It should be some "min cruise throttle" to avoid stall in tail wind but the target cruise Speed configurable to go full power if needed when Fighting against the wind?

also it would be nice to get clearer Information on the advanced Tuning tab what Settings are Active for fixed wing or copter only and what are active for both.

Tygrys-1, 2018-09-09:
I think that in automated modes the (cruise) throttle can vary from Min. throttle to Max. throttle. But I agree that the logic for this should be more advanced.

The problem with the correct cruise speed setting is that stalling is dependent on airspeed, and airspeed sensor is not that common. GPS give only ground speed which is not very usefull in this scenario.

The nav_fw_cruise_speed is a parameter to estimate flight range/time at cruise speed, not for driving the plane - so it is not usefull that much.

Find more about it in [Wiki](https://github.com/iNavFlight/inav/blob/master/docs/Cli.md)

shellixyz, 2018-09-10:
As stated by @Tygrys-1 `nav_fw_cruise_speed` is not used by the navigation system, it is an estimation of the speed when the aircraft is cruising during which the mean motor command should be around `nav_fw_cruise_thr`. It is used to calculate the remaining flight time and distance before you need to RTH which you can display on the OSD.

A hardcoded minimum ground speed is already enforced. If you're flying against the wind the navigation system will increase the throttle up to `nav_fw_max_thr` so that you're always flying at least at 6.5m/s. The throttle is increased by (700 - speed_in_cm_per_s) * 1.5 per second if the ground speed is bellow 6.5 m/s.

b14ckyy, 2019-02-21:
Hi @shellixyz 

I have worked very intensive with INAV during the last months and also learned a lot about the different CLI values that are helpful and not in the configurator GUI. 

So now I have a good list of all values for fixed wing that, in my opinion, are really important for a good first time config and should be accessible easy also for beginners:

Highest Prio:
nav_fw_launch_max_angle 
nav_fw_launch_thr
nav_fw_launch_climb_angle
nav_fw_cruise_yaw_rate
nav_fw_allow_manual_thr_increase

Not as important but helpful:
nav_fw_launch_motor_delay
nav_fw_launch_spinup_time

IMHO all these values MUST be configured before first flight, especially the Autolaunch features. I read a lot of pilots reporting their plaines crashed into the ground because of the low default climb angle and the too low launch angle. Especially for smaller and powerful wings, these should be set much higher.

shellixyz, 2019-02-21:
You're right it would be better if these settings were available through the GUI. I agree about the max launch angle, it could be set higher, but I don't agree about the climb angle. Personally I always set the climb angle lower than the default 18°.

b14ckyy, 2019-02-21:
I had to make the experience, that starting with a low launch angle can cause hard nose dives. Because the accelleration of the plane will lead to a slight backwards g-force on the accellerometer and the FC "thinks" the plane is already climbing and possibly it detcts a higher climb angle and pitches further down. 

Even with my 45° I have set it first goes nearly flat for 1-2s while it is accellerating at full throttle and then pitches up to a clean and straight climb. 

But anyway. I think this option should be in the configurator and anybody can decide for themselve :)

MrD-RC, 2020-10-29:
A lot of time has passed since this was requested. Going through the latest configurator (as of 29/10/2020), I can see the following are still missing. I'll look to add them.

nav_wp_radius
nav_wp_safe_distance

nav_fw_cruise_yaw_rate
nav_fw_allow_manual_thr_increase

I'd also like to see loiter direction added.

avsaase, 2021-04-03:
Seems all the requested settings have been implemented in the GUI.

closed 2021-04-03 by avsaase