iNavFlight/inav-configurator #571: v2.0 mixer -> missing RC Channel 9 to 16, and missing servo 8 and 9

davidngrc, 2018-09-07:
in the iNav Configurator 2.0 Final

1) the RC channel 9 to 16 still missing in the mixer tab.
this was raise in https://github.com/iNavFlight/inav-configurator/issues/501
but no fix.

in CLI,   smix 4 7 11 100 0
that 11 is RC Channel 8, so I assume channel 9 will be 12
then I do     smix 4 7 12 100 0
it show as "Gimbal Pitch"
where is my rc channel 9 to 16, come back please...

2) some FC have 9 servo, like matek f405 wing, in the "servo mixer“ section, only 7 servo is allowed to select.
please add 2 more servo...

3) I try to debug this, but I am using windows, when add source code to chrome, the extension UI has no graphic.
in the README.md it said: With Chrome: Run `./node_modules/gulp/bin/gulp.js`. Then open `chrome://extensions`, enable
I try to run the gulpfile.js in note.js command prompt, it shows some error.
I am stuck... please help...

[image]

[image]

Arvycka, 2018-09-20:
This issue is mysteriously ignored for some reason :(
It's been raised for the second time and not a single coment.

Arvycka, 2018-10-02:
So what's up with that? It was supposed to be like:
"INAV 2.0 will free mixer settings and it will be possible to mix any RC channel into any servo on any airplane, so POT will be able to drive flap throw and so."

Instead the new mixer ir very limited :(

Arvycka, 2018-10-02:
You need to start with servo "2" in mixer. Servo 2 will be connected on S3. Servo3 - S4 etc.
Then S7 and S8 (servos 5 and 6)will be available for other functions.
It's still impossible to use S9 (servo 7)
RC channels 9-16 missing from the mixer is really limiting and annoying :(

DzikuVx, 2018-10-07:
INAV is currently supporting 8 servos max. Indexed from 0 to 7. This is not FC dependant at all

DzikuVx, 2018-10-07:
fixed in #581

PR #581 (2018-10-07): Rc channels 9-16 added to mixer GUI
fixes #571
closed 2018-10-07 by DzikuVx