iNavFlight/inav-configurator #583: Cannot configure servos in iNav 1.9.1 with 2.0 configurator

Tygrys-1, 2018-10-08:
The servos tab is empty. It tells to configure the mixer.
The mixer tab says that it is not supported on 1.9.1.

Can the configurator be smarter and allow servo page to work as before on pre 2.0 software?

davidngrc, 2018-10-09:
why not use the v1.9.1 configurator? https://github.com/iNavFlight/inav-configurator/releases/tag/1.9.1
v2.0 change so many things, I don't think the developer have time to do backward compatibility test.

Tygrys-1, 2018-10-11:
Because when I start the configurator 1.9.3 it told me that there is a new version to upgrade. So I upgraded :-)

DzikuVx, 2019-01-21:
This land in "no fix" zone I'm affraid. With Configurator 2.1 we lock firmwares older than 2.0

closed 2019-01-21 by DzikuVx