iNavFlight/inav-configurator #593: add external wms map support to configurator

tednv, 2018-11-09:
## Current Behavior
iNav GPS and Mission Control panes currently use one online map provider

## Desired Behavior
Allow iNav to change the type of map you're using or use a map proxy

## Suggested Solution
Allow selection of any WMS map provider (or any other protocol, WMS is easiest to implement). There are several WMS proxies available for offline caching

## Who does this impact? Who is this for?
This will allow the use of maps while offline or in an area that does not have cell phone signal or internet connectivity, or different types of maps depending on one's preference

## Additional context
https://wiki.openstreetmap.org/wiki/MapProxy

There will be no legal issues with this implementation since it will be up to the end user to setup their own offline caches with whatever tiles they want

tednv, 2018-11-27:
adding support for mapproxy #593 #603

DzikuVx, 2019-01-04:
closing as done in #603

closed 2019-01-04 by DzikuVx