iNavFlight/inav-configurator #597: Nav poshold problem

Emporiodroni, 2018-11-11:
[blackbox_log_2018-11-11_172439.TXT](https://github.com/iNavFlight/inav-configurator/files/2569567/blackbox_log_2018-11-11_172439.TXT)
Hi, I'm having a problem to activate NAV POSHOLD / HEADING HOLD function. I'm using an SP F3 FC with NAZA GPS on a 250 size frame. I have 9-10 satellites, the quad is flying OK in angle mode (besides of a mild backwards drift) but when I activate with a switch NAV POSHOLD nothing happens, the quad is not keeping the position or not even changing the flying behaviour. I have calibrated accelerometer, compass, no warning signals. I have tried NAV POSHOLD alone and also together with HEADING HOLD with the same switch with no luck. Please help me to find the solution to this problem.

I have attached the some pictures, dump file and blackbox file. 
[image]
[image]
[DR0111_150_Flysky_Inav_F3_NAZA!.txt](https://github.com/iNavFlight/inav-configurator/files/2569563/DR0111_150_Flysky_Inav_F3_NAZA.txt)

[image]

Castion01, 2018-11-17:
I've made some more flying tests. Programmed a switch pos1: Angle mode, pos2: Heading hold & Nav Poshold (and NO Angle mode). I armed the quad in pos1 (Angle mode) then I was hovering around in 1m altitude then moved the flight mode switch to pos2. The quad did not keep the position but was not even in Angle mode which I found a bit odd as Nav Poshold mode is by default combined with Angle mode. 

Does it mean that the Nav poshold was not active (despite that flight mode was actived with the switch)?

kardon18, 2018-12-11:
Are you aware that **Nav Poshold** is a Loiter function? So the quad will take up a stationary position in the air.  
It is best used in conjunction with **Altitude hold** mode.  This will give the quad the ability to hold, not only 2D position, but altitude as well.
You need a barometer for this to work. And I noticed from your configurator screen shoots, that the Barometer is not configured or connected correctly.

shellixyz, 2018-12-11:
@Castion01 Sounds like the PH mode wasn't activating. It can happen even if you have a 3D fix when the estimated position precision is too low. For now if the PH mode is not activating and you don't have any other mode combined to it it is like you have no mode selected (ACRO). Have you got a BB log ?

@kardon18 Since 2.0 PH is always 3D you don't need to add AH to it.

Castion01, 2018-12-15:
Thanks for the advises so far. I have activated the PH in the air.
@shellixyz, In theory, the PH mode by-default including the Angle mode (https://github.com/iNavFlight/inav/wiki/Navigation-modes). But I have tried also with Angle mode with no success. 
My supposition is that F3 Acro and NAZA GPS somehow are not compatible. I've ordered a M8N GPS to do other tests with the same Betaflight setup.

Castion01, 2019-01-03:
Looks like I've found the problem and it was the sbus setup. Changed to PPM and now works.

closed 2019-02-10 by DzikuVx

davidbitton, 2019-03-24:
@Castion01 any idea why that might be?

Aaronprat, 2019-04-18:
@Castion01 can you share more information about the solution?

Thanks

fremebondo, 2019-07-02:
Hi, I am facing the same exact issue on a fixed wing. Any hint to solve? I'm using SBus to read from taranis X8R receiver.

Palput, 2021-03-16:
hello, I am interested in this case and I am researching all about it, i would appreciate information. 
Thank you
Pd: as soon as I change PPM I will say if it really works for future reference.