iNavFlight/inav-configurator #605: Vision Font - Additions for increased sharpness and clarity

kardon18, 2018-12-02:
Vision font is my most preferred font to use. Although I do find that some of the characters are a little difficult to see and make out on lower resolution screens.
I went through and changed the highlights a layout of some of the characters, so they appear clearer to read at a glance. And look a little sharper.

If they seem acceptable. Would you mind adding some of them?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

shellixyz, 2018-12-02:
Could you upload a zip of the directory with the modified chars please ? Would make comparing the old and new chars much easier.

kardon18, 2018-12-03:
I wasn't sure if you meant a zip folder of the .png  

Or was talking about adding a zip with the modified characters being ran through the Max7456tool and converted to .mcm 
That is a bit over my head at the moment. I'm trying to work out the Max7456tool. 

[Font modifications.zip](https://github.com/iNavFlight/inav-configurator/files/2637616/Font.modifications.zip)

shellixyz, 2018-12-03:
Perfect, it is what I needed. Not everybody likes your modifications so I proposed them as a new font "Vision Alt". See #608.

kardon18, 2018-12-04:
@shellixyz -Not everybody likes your modifications so I proposed them as a new font "Vision Alt"

That sounds good to me.
I wasn't expecting that you would propose a new font. But to my surprise it was added. Thanks Michel.

I only changed a few thing, hoping they may be adapted to the current Vision font. 
But due to it being its own font. I have given it some more individuality. And have tried to make it stand by itself.
There a few new extra's I worked on today. I did this after using the version you made last night. And flying with it today, to see if any area's needed cleaning up.
It looks okay. But I though I may as well go the whole way before its release in 2.1.

I know you went to a lot of effort yesterday to get it all going. 

Would you mind again incorporating these into Vision-Alt.
(I also added the new 3D speed symbols in this lot too)

Thanks

[vision-alt (tweeks).zip](https://github.com/iNavFlight/inav-configurator/files/2643518/vision-alt.tweeks.zip)

shellixyz, 2018-12-04:
> Would you mind again incorporating these into Vision-Alt.

No problem. I just updated the PR with your new proposed modifications :)

kardon18, 2018-12-04:
Thanks again Michel for your help.

I took it out this morning with the latest editions. And i'm very pleased with the appearance. 
I was concerned about how the variant of the new home arrow, would look. It took some getting right, on certain angles. But it work well. It has a shallower look to it. More like the StarTrek/Starfleet insignia.

[image]
[image]

kardon18, 2018-12-04:
Thanks again Michel for your help.

I took it out this morning with the latest editions. And i'm very pleased with the appearance.
I was concerned about how the variant of the new home arrow, would look. It took some getting right, on certain angles. But it works well. It has a shallower look to it. More like the StarTrek/Starfleet insignia.

[image]
[image]

kardon18, 2018-12-05:
With these it should be finished. I noticed that Speed and 3D Speed were different in layout.

Let me know of 152 "gyro". I added two others in this zip. Choose which one you like best of these.  Or just leave the one from yesterday.

[vision-alt fixes.zip](https://github.com/iNavFlight/inav-configurator/files/2649643/vision-alt.fixes.zip)

shellixyz, 2018-12-05:
I updated the PR. Symbol 152 doesn't matter, it is not used.

kardon18, 2018-12-09:
I did a lot of flying yesterday, with the vision-alt.  And noticed that a some of the icons needed toning down a little on their boarders. Just to keep in line with vision fonts over all look. 
Although these new tones still leave it more highlighted than the original vision. Which is the look I was after. Not too much...Not too little.
I think I'm becoming OCD :/
Thanks again.

[vision-alt (more tweeks).zip](https://github.com/iNavFlight/inav-configurator/files/2660197/vision-alt.more.tweeks.zip)

shellixyz, 2018-12-09:
Hehehehe ;)

PR Updated

shellixyz, 2018-12-17:
PR merged, closing

kardon18, 2018-12-17:
Hi Michel,
Sorry to be a pain.
I didn't know when you would merge the font.
And I have been fine tuning it, and adding many extra's. With some flight buddies trying it too. And giving me their impressions.
It now has some modified letters and numbers as well. Most of vision has been altered for the new look of vision-alt.
It also 3D home and wind direction indicators.
Let me know what you think?

**It is now finished and read to merge.**
Thanks for your patience with me, as I was getting it spot on.
  
[image]

The files is already in .MCM
[vision-alt.zip](https://github.com/iNavFlight/inav-configurator/files/2688077/vision-alt.zip)

shellixyz, 2018-12-17:
Sorry I should have asked you. Looking good ! I will update it.

kardon18, 2018-12-29:
@shellixyz 
I have another dozen updates to the vision-alt font. Which changes its appearance a little more, in area's.
I won't post them here yet. Because I've noticed that there have been a few extra functions added to inav, that may require glyph additions or edits. Before 2.1 is released.

Keep me informed of whether the font needs any additions. And when the time comes for the 2.1 RC.
Then i'll post the rest.

Thanks

shellixyz, 2019-01-03:
Sorry, been away for a week. I'm slowly going through the GH mails. 

I don't see any glyph needing to be changed or added for the new features. The only feature I can think of that needs new glyphs in 2.1 is the 360° AHI but the glyphs for that (vertical horizon bars) are already present.

kardon18, 2019-01-06:
I hadn't heard back from you Shelli.
I wasn't sure if you were working on the above mentioned or not. So I thought I'd post the finished Vision-alt font anyway.

[vision-alt(finished).zip](https://github.com/iNavFlight/inav-configurator/files/2730214/vision-alt.finished.zip)

shellixyz, 2019-01-06:
Sorry, looks like I've missed the notification mail. You last update looks good. I didn't include the symbols 8 to 12 in the update because they are unused. I can include your propositions for the remaining flight time and distance symbols, they are looking good. We're going to release 2.1RC very soon.

kardon18, 2019-01-07:
Thanks for the update :)

The file here includes :
1) Flight time remaining in - hours / minutes
2) Distance remaining in - Km / Meters / Miles / Feet  

There are six glyphs in total, for each font.
With each gylph having its own characteristics style for that font.

I know six extra symbols for this function is a lot. But I figured; Unless the Remaining time in HOURS was included. A bug might turn up.

I thought this my save you some time. 
[Remaining time_Remaining distance_glyphs.zip](https://github.com/iNavFlight/inav-configurator/files/2731024/Remaining.time_Remaining.distance_glyphs.zip)

kardon18, 2019-01-07:
How about this symbol  for #3587

[image]
[Over home position icon.zip](https://github.com/iNavFlight/inav-configurator/files/2731114/Over.home.position.icon.zip)

shellixyz, 2019-01-07:
Nice. I didn't check how it looks on the OSD but it should do the job :+1:

shellixyz, 2019-01-08:
We're going to wait for 2.2 to decide if we're adding new symbols for the remaining distance and time because of the new changes to the OSD code in 2.1

closed 2019-01-08 by shellixyz

kardon18, 2019-01-09:
Opps.
I didn't mean to step on anyone's toes.

How about "IMPACT" font.  
Its similar to that style.

kardon18, 2019-01-12:
Hi Shelli,
I noticed a few extra commit features.  **Home Slope and Total** 

@danarrib  and @OlivierC-FR have added fonts in Clarity and Vision accordingly.

I thought i'd do the same for Impact.
Plus I also did another "Over Home Location" symbol that is 3D in appearance.  Its still based on the above, glyph I posted; But it better suits the style of the 3D home arrow in Impact.

[Home_Total and Slope glyphs for Impact font.zip](https://github.com/iNavFlight/inav-configurator/files/2751696/Home_Total.and.Slope.glyphs.for.Impact.font.zip)

Let me know what you think.
[image]

kardon18, 2019-01-21:
> 
> 
> It's looking good ! I will take the time this weekend to add the code for the home glyph and update the font.

Thanks Shelli

kardon18, 2019-01-30:
Hi Shelli, Any chance of you integrating the above characters, plus a few that I fixed up by request. Before the full version of 2.1 is released?  
All are in this zip.
Thank you
[Update with slope_home_total.zip](https://github.com/iNavFlight/inav-configurator/files/2810733/Update.with.slope_home_total.zip)

kardon18, 2019-02-08:
Thank you Alberto :+1:  Much appreciated.

Jetrell, 2019-06-11:
Hey guys, I found this original Impact font thread.
Over the months I have used Impact font with inav and cleaned it up.
This upgrade/fix/addition adds better 3D looking home arrows, with other cleanups.

Any chance @fiam or @shellixyz of adding them?
[Impact 2.2 additions.zip](https://github.com/iNavFlight/inav-configurator/files/3275365/Impact.2.2.additions.zip)