iNavFlight/inav-configurator #606: Failure during reboot

Cyremiaepf, 2018-12-02:
Hello,

I'm actually working on a MATEK f405-STD and i have an issue when i plug it on inav. Every time i try to save a new preset and i need to reboot, there is a failure during the process:

EEPROM saved

2018-12-02 @ 15:51:05 -- Device - Rebooting

2018-12-02 @ 15:51:06 -- Unrecoverable failure of serial connection, disconnecting...

2018-12-02 @ 15:51:06 -- Failed to close serial port

I used zadig and Impulse RC driver to set the drivers.

Any ideas of how to fix it ?

Thanks

uCibar, 2018-12-12:
My flight controller is MATEK F405-WING and i have same issue too.  INAV Configurator is 2.0.0. and Firmware is MATEK405SE 2.0.1. Did you fix this @Cyremiaepf ?

Cyremiaepf, 2018-12-12:
Hi,

not yet i still have the same issue.....
But for the moment it takes into account the settings i choose, however there is this message.

If you find a solution tell me !