iNavFlight/inav-configurator #619: LED Strip Non-Functional 2.0.0

teckel12, 2018-12-22:
When I go into the LED Strip page, it shows "0 remaining" even though I haven't programmed any LEDs (new FC).  Also, there doesn't seem to be any LED config data when I do a dump (2.1 pre-release).

Has LED functionality been totally removed?  I know it never really worked well, but at least I could get the LEDs to light and change to red when the battery was low.

shellixyz, 2018-12-22:
It seems to be working fine for me (master branch + OMNIBUS). Maybe the target you're using doesn't have the LED_STRIP feature enabled.

[image]

teckel12, 2018-12-22:
@shellixyz How would the configurator know if my target supports LED strip? My target is new, so 2.0.0 doesn't know it even exists.

I haven't messed with the configurator source, so I really don't know how it operates. But it allows me to activate LED strip and LED strip is on the menu.  Maybe this is what happens with a target that the configurator doesn't know about? 

I'll close this and take it to Telegram as a question instead of an issue.

teckel12, 2018-12-22:
The target MATEKF722SE has LED and buzzer enabled, but it seems the configurator doesn't know about it.

teckel12, 2018-12-23:
Closing this as it's got to be an issue with the target

closed 2018-12-23 by teckel12