iNavFlight/inav-configurator #622: cannot run npm install on raspberry pi

wx4cb, 2018-12-27:
charlesb@GroundStation:/usr/src/inav-configurator $ npm install
[ .................] | preinstall:inav-configurator: info lifecycle inav-configurator@2.0.0~preinstall: inav-configurator@2.0.0

> uws@9.14.0 install /usr/src/inav-configurator/node_modules/uws
> node-gyp rebuild > build_log.txt 2>&1 || exit 0

> openlayers@4.6.5 install /usr/src/inav-configurator/node_modules/openlayers
> node tasks/install.js

> closure-util@1.26.0 postinstall /usr/src/inav-configurator/node_modules/closure-util
> node ./bin/closure-util.js update

info install Downloading http://dl.google.com/closure-compiler/compiler-20171112.zip
info install Downloading http://github.com/google/closure-library/archive/v20170910.zip
info install Download complete: /usr/src/inav-configurator/node_modules/closure-util/.deps/compiler/5eb2c411b449e9f3d879be0f59139cb5b2894c09
info install Download complete: /usr/src/inav-configurator/node_modules/closure-util/.deps/library/947e46b02fd5e472c2daa86669c09303ca3e7033

> nw@0.31.4-sdk postinstall /usr/src/inav-configurator/node_modules/nw
> node scripts/install.js

Response code 404 (Not Found)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nw@0.31.4-sdk postinstall: `node scripts/install.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the nw@0.31.4-sdk postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/charlesb/.npm/_logs/2018-12-27T18_15_36_878Z-debug.log

nmaggioni, 2018-12-28:
Have you tried running `npm install` once again after having deleted the `node_modules` folder? I've just had the same problem, it looked like being cache-related to me.

wx4cb, 2019-03-06:
i've done all that. fresh ubuntu install (not raspbian as that has other issues), I now get the following after the git clone and installing npm from apt:

```
charlesb@GroundStation:~/src/inav/inav-configurator$ npm install
npm WARN deprecated openlayers@4.6.5: Please use the ol package instead
npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.
npm WARN deprecated cryptiles@2.0.5: This version is no longer maintained. Please upgrade to the latest version.
npm WARN deprecated boom@2.10.1: This version is no longer maintained. Please upgrade to the latest version.
npm WARN deprecated hoek@2.16.3: This version is no longer maintained. Please upgrade to the latest version.
npm WARN deprecated uws@9.14.0: New code is available at github.com/uNetworking/uWebSockets.js

> uws@9.14.0 install /home/charlesb/src/inav/inav-configurator/node_modules/uws
> node-gyp rebuild > build_log.txt 2>&1 || exit 0

charlesb@GroundStation:~/src/inav/inav-configurator$ npm start

> inav-configurator@2.1.4 start /home/charlesb/src/inav/inav-configurator
> node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .

[19:40:50] Using gulpfile ~/src/inav/inav-configurator/gulpfile.js
[19:40:50] Starting 'build'...
[19:40:50] Starting 'build-all-css'...
[19:40:50] Starting 'build-all-js'...
[19:40:50] Starting 'build-css'...
[19:40:50] Starting 'build-receiver-css'...
[19:40:50] Starting 'build-js'...
[19:40:50] Starting 'build-receiver-js'...
[19:40:50] Starting 'build-debug-trace-js'...
[19:40:50] Starting 'build-hex-parser-js'...
[19:40:51] Finished 'build-debug-trace-js' after 681 ms
[19:40:51] Finished 'build-hex-parser-js' after 685 ms
[19:40:51] Finished 'build-receiver-css' after 750 ms
[19:40:51] Finished 'build-receiver-js' after 763 ms
[19:40:51] Finished 'build-css' after 1 s
[19:40:51] Finished 'build-all-css' after 1.01 s
[19:40:52] Finished 'build-js' after 1.25 s
[19:40:52] Finished 'build-all-js' after 1.26 s
[19:40:52] Finished 'build' after 1.27 s
nw.js appears to have failed to download and extract. Attempting to download and extract again...
Response code 404 (Not Found)
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: spawn /home/charlesb/src/inav/inav-configurator/node_modules/nw/nwjs/nw ENOENT
    at _errnoException (util.js:1022:11)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:190:19)
    at onErrorNT (internal/child_process.js:372:16)
    at _combinedTickCallback (internal/process/next_tick.js:138:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)

npm ERR! Linux 4.15.0-1031-raspi2
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! inav-configurator@2.1.4 start: `node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the inav-configurator@2.1.4 start script 'node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the inav-configurator package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs inav-configurator
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls inav-configurator
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/charlesb/src/inav/inav-configurator/npm-debug.log
```

wx4cb, 2019-03-06:
and as far as that log file

```
charlesb@GroundStation:~/src/inav/inav-configurator$ cat /home/charlesb/src/inav/inav-configurator/npm-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@3.5.2
3 info using node@v8.10.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle inav-configurator@2.1.4~prestart: inav-configurator@2.1.4
6 silly lifecycle inav-configurator@2.1.4~prestart: no script for prestart, continuing
7 info lifecycle inav-configurator@2.1.4~start: inav-configurator@2.1.4
8 verbose lifecycle inav-configurator@2.1.4~start: unsafe-perm in lifecycle true
9 verbose lifecycle inav-configurator@2.1.4~start: PATH: /usr/share/npm/bin/node-gyp-bin:/home/charlesb/src/inav/inav-configurator/node_modules/.bin:/home/charlesb/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/charlesb/bin
10 verbose lifecycle inav-configurator@2.1.4~start: CWD: /home/charlesb/src/inav/inav-configurator
11 silly lifecycle inav-configurator@2.1.4~start: Args: [ '-c',
11 silly lifecycle   'node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .' ]
12 silly lifecycle inav-configurator@2.1.4~start: Returned: code: 1  signal: null
13 info lifecycle inav-configurator@2.1.4~start: Failed to exec start script
14 verbose stack Error: inav-configurator@2.1.4 start: `node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/share/npm/lib/utils/lifecycle.js:232:16)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at EventEmitter.emit (events.js:214:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/share/npm/lib/utils/spawn.js:24:14)
14 verbose stack     at emitTwo (events.js:126:13)
14 verbose stack     at ChildProcess.emit (events.js:214:7)
14 verbose stack     at maybeClose (internal/child_process.js:925:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
15 verbose pkgid inav-configurator@2.1.4
16 verbose cwd /home/charlesb/src/inav/inav-configurator
17 error Linux 4.15.0-1031-raspi2
18 error argv "/usr/bin/node" "/usr/bin/npm" "start"
19 error node v8.10.0
20 error npm  v3.5.2
21 error code ELIFECYCLE
22 error inav-configurator@2.1.4 start: `node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .`
22 error Exit status 1
23 error Failed at the inav-configurator@2.1.4 start script 'node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the inav-configurator package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     node_modules/gulp/bin/gulp.js build && node_modules/nw/bin/nw .
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs inav-configurator
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls inav-configurator
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

closed 2019-05-12 by DzikuVx