iNavFlight/inav-configurator #648: 2.1 Failure to restore a full backup

OlivierC-FR, 2019-01-11:
- Flash a fresh iNav 2.1.RC1
- Configure it all
- CLI dump, save to a file
- Reflash the board again, with same RC1, or whatever 2.1 dev version
- Paste the full dump from the TXT file in the CLI, save
- Restart it all, dump the CLI again, paste into a new file
- Compare both files : Only about the first 70-80% of the settings are saved.

Consistent over 2 different PCs (Window 10 Laptop, Windows 10 Desktop) and 4 flight controllers : Matek F405-CTR x 2, F405-wing, F405-CTD

These screencaps for one particular PC/aircraft combo, but it's doing the same with rest of the gear.
I don't recall if it was cutting at the exact same place, but was very similar, about 20-30% of the last commands are not sent when the dump is pasted.

For this sample it skips 140 lines, starting at line 818 :

[image]

[image]

kardon18, 2019-01-12:
Did you try to paste `DIFF ALL` files, instead of `DUMP`?

OlivierC-FR, 2019-01-12:
No, but that's not really the point, the point is to make a full backup :)

kardon18, 2019-01-12:
@OlivierC-FR 
I understand. 
That's what I was getting at. Whether the error would still occur, if the backup was pasted in small blocks.

I have noticed that 2.1.0 seems to write the backup files much faster that previous versions did. By just watching how fast the written settings go down the CLI.  Have you noticed that?

I remember @stronnag  mentioning the possibility of a compiler error occurring during file dumps; on RCG once. 
That's why I only save DIFF ALL back ups.

OlivierC-FR, 2019-01-13:
Yes its a lot faster to paste now, maybe too fast for its own good !

I made another flash this morning, different FC (Matek F450-WING), different PC (my Win10 laptop instead of min Win10 desktop).
Also fails to paste the whole dump, just this time it's only the very end.

[image]
[image]

kardon18, 2019-01-13:
Hopefully one of the guys pick up on this and gets to the bottom of it. Before it misses loading a more critical file, for someone who wouldn't think to look for a write error.

I've got into the habbit of checking through my loaded files before I fly. And after seeing how fast it compiled the settings now. And after you having this issue. I'll be watching more, until a fix is released.

My writes are only DIFFs. And I've had no errors as yet.

Does there seem to be any pattern of specific settings?

DzikuVx, 2019-01-23:
if anyone can test #661 that would be awesome

DzikuVx, 2019-01-23:
#661 is fixing the issue

closed 2019-01-23 by DzikuVx