iNavFlight/inav-configurator #649: Disable support for DSHOT600 and DSHOT1200 for reliability reasons

digitalentity, 2019-01-12:
Per issue title.
Summoning @DzikuVx @fiam @stronnag @shellixyz for discussing this

digitalentity, 2019-01-12:
I wonder if we should add the same warning regarding high looptime frequencies. 4-8KHz looptime is unsustainable if you want navigation capabilities.

DzikuVx, 2019-01-12:
@teckel12 yes, DSHOT600, Matek F722-STD, looptime 4K and everything was fine.

DzikuVx, 2019-01-12:
@digitalentity I think something can be arranged for high looptime as well

Vlad1384, 2019-01-12:
Matek F722-STD DSHOT1200 died Matek FCHUB-6S. As soon as I put the DSHOT1200 and saved the settings right there instead of 5 and 10 volts, it became 3 volts everywhere. I do not know what it is connected with, but before that I flew on multishot everything worked normally.

Vlad1384, 2019-01-12:
Well, if so. I will recover ...

teckel12, 2019-01-12:
@Nihilistic I'm running multishot on my F722SE for at least the next year. I see no advantage of using DSHOT and only potential catastrophic failures. If you use DSHOT, fully expect a problem, hence the warning.

Nihilistic, 2019-01-12:
@teckel12,

So, are we looking at hardware issues with this board? Or is there...shall we say...vast room for improvement with the iNAV target for this board?

Nihilistic, 2019-01-13:
@teckel12,

I may give DSHOT 150 and/or 300 a try, and just stay in close and below 50 feet. And if I can get through a dozen flights or so without incident...it might build some confidence in it. I don't blame you for not wanting to risk smashing or losing your babies ;)

But, the devs seem to think there's nothing wrong, because they've been flying DSHOT for a while (probably not on F7's). So, if nobody is willing to test and report their results, how is it going to improve?

teckel12, 2019-01-13:
@Nihilistic There's vast room for improvement in the DSHOT protocol.

digitalentity, 2019-01-13:
This issue is not about F3 vs F4 vs F7 targets or any specific boards. It's about inherent unsafety of DSHOT protocol. For fundamental reasons the requirements for wiring quality for DSHOT150 are roughly the same as for Multishot. This means that if you want to use DSHOT and have similar reliability you have to invest into better wiring.

DSHOT has some room for improvement, but until people understand that it requires much higher quality builds than a bunch of tangled wires we need to warn them about it's reduced reliability.

Nihilistic, 2019-01-13:
I'm using individual ESC's on the arms, and the signal wires are directly soldered to the signal pads at the corners of the F722-SE. They're twisted around the ground wires, which are soldered to the same pad as the big ground pad for each ESC respectively. The wire length is maybe 2 inches. The solders are good clean shiny joints. The very same ESCs were soldered to a Realacc PDB in the same way, with short signal wires soldered to it, which were in turn soldered to the Omnibus F3, loaded with BetaFlight 3.5.X, and ran DSHOT600 successfully for many flights, without incident. I really don't think it's my wiring or soldering.

Nihilistic, 2019-01-13:
The wires are about as short as they can be. You don't want them to be taut, because they'll be more likely to transmit vibrations to the board's gyro, and that would moot the use of vibration dampening bobbins the board is mounted on.

Nihilistic, 2019-01-13:
I'll try DSHOT300 and see what happens, but there's snow on the ground here now, and probably will be for a while. My quads aren't water proofed, so I don't want crash landings into the snow.

digitalentity, 2019-01-13:
There are people who have made a lot of flights with DSHOT on INAV as well, also without issue. Until we have logging (or at least error reporting) capabilities on ESC side we can't conclude anything. 

It's not about wire length. Bandwidth required for DSHOT 600 is about 6-7 MHz, which is well into radio frequency range. Without external noise a telecom-grade twisted pair (cat3+) could be used for those kinds of signals, but in noisy systems to get anywhere near being reliable you need a coaxial cable.

shellixyz, 2019-01-13:
I agree a warning is better than disabling 600 & 1200. BTW at least with BLHeli32 (maybe with BLHeli_S I didn't check) it is possible to see if there are dropped frames and the rate by connecting to the ESCs with BLHeli suite. The number of frames / dropped is shown in the setup summary window that appears right after enumerating the ESCs.

digitalentity, 2019-01-13:
@shellixyz not many of us keep quad powered after a crash for long enough to connect to BLHeli and figure things out. I once asked BLHeli devs if they would expose that data via telemetry and they said no, unfortunately.

DzikuVx, 2019-01-15:
done in #652

closed 2019-01-15 by DzikuVx