iNavFlight/inav-configurator #655: [Feature Request] See GPS data in GPS tab

wx4cb, 2019-01-17:
is it possible to get the raw NMEA sentences shown at the bottom of the GPS tab? 

I know there's the message count, but not the actual messages. It would be nice if we could get that data scrolling at the bottom so you can correlate it if you are getting weird data. maybe add it as a switch on/off kinda like the live mode in the servos tab.

KiwiDavid, 2019-02-02:
It won't be NMEA data as you might see with u-center and a default GPS module set up.  At power up iNav will reconfigure the GPS to output ubx (u-blox binary) and I don't believe the more readable NMEA is used.
Perhaps the switch you mention could be used to keep it in NMEA mode and leave the update rate at 1Hz so it's more easily read.   I agree it could be useful for diagnostic work such as tracking down noisy electronics.

wx4cb, 2019-02-02:
@KiwiDavid ahh makes sens.... but yes switch it to nmea for testing would still be a bonus. I'm used to reading NMEA raw so for me i can easily parse most of it in my head - hence the request.

shellixyz, 2019-03-24:
It could be done for NMEA I guess but most people use UBlox modules which are using a binary protocol. IMO this won't be very useful.

MrD-RC, 2022-11-26:
NMEA is deprecated in 6.0, and will be removed soon after.

closed 2022-11-26 by MrD-RC