iNavFlight/inav-configurator #659: INAV Configurator 2.1.1   -

JVN2000, 2019-01-21:
Cannot configure OSD with new configurator - Blank Screen
INAV, version: 2.0.1 - MATEK F411-Wing

gbine, 2019-01-22:
Confirm same issue with Matek F405-Wing

kardon18, 2019-01-23:
I found the same thing happen.
The issue was also there in RC1. An attempt was made for a fix in RC2 . But the issue still remains.

The configurator can be closed and restarted, with the FC also being rebooted. Then the OSD page on the configurator will unlock, when Java script reloads.

jreise-d, 2019-01-23:
Same here, blank OSD page in 2.1.1, regardless of inav version on my f411 wing. I can reproduce this on several computers with win7 or win10, if no inav configurator has run on it yet, or folder users\xxx\AppData\Local\inav-configurator doesn't exit.

My temporary solution is this: I start configurator **version 2.1.0**, connect f411 and open OSD page, everything seems to be ok. Then I try 2.1.1 and everything is ok, OSD page is working. It seems like version 2.1.0 is writing something into the folder \AppData\Local\inav-configurator that version 2.1.1 needs for proper function.

noagendafpv, 2019-01-24:
No OSD issues for me, Matek F411-wing, loaded iNav 2.0.1 and the 2.1.1 config would not communicate. Tried with a second F411 and same result, I reverted to config 2.1.0 and no more problems.

putimir, 2019-01-24:
Same here...

+: https://www.facebook.com/groups/1309125302467027/permalink/2074959035883646/

putimir, 2019-01-24:
@JVN2000 can you rephrase issue title to be more descriptive, please

machadofelipe, 2019-01-24:
Same issue here, both on Win10 and MacOS (both with Chrome version 71.0.3578.98 (Official Build) (64-bit)). I was coming from 2.0, and after installing 2.1.1, OSD page opens blank. By reverting back to 2.1.0, opening the OSD tab and them installing the 2.1.1 solves de issue.

noagendafpv, 2019-01-24:
> No OSD issues for me, Matek F411-wing, loaded iNav 2.0.1 and the 2.1.1 config would not communicate. Tried with a second F411 and same result, I reverted to config 2.1.0 and no more problems.

I can actually change my above statement, the OSD tab seems to be what is locking up my configurator. I have gone back to 2.1.0 though and it seems to work better.

MCGUIRKINATOR, 2019-01-25:
Still experencing the same issue with osd tab blank on 2.0.1 and 2.1.1

Kennymartin, 2019-01-26:
same osd tab not working with latest configurator!  Any  fixes yet???????

MCGUIRKINATOR, 2019-01-26:
I reflasehed a few times to 2.1.1 with a kakute f4 v2 and got mostly every
thing working gps compas baro and osd tab, but something  went wrong with
the i2c bus and baro and mag come and go and esc would not arm regardless
of protocol went back to betaflight got gps working but no mag or baro.
Must be a faulty board

MCGUIRKINATOR, 2019-01-26:
> Sounds like your board was faulty alright.
> I think the guys at inav have not finalised the latest version of
> configurator to work fully with the latest firmware on the boards.
> My setup is finished, everything is working as it should except the
> ability to adjust my OSD via the configurator tab and I'm just waiting on
> good weather to maiden her.
> Guess I'll have to live with the default OSD till they get around to it!

Have you tried flashing your board to 2.1.0 with hex file from git hub that
had osd working. It took about 30 seconds to load the tab but I was able to
configure it.

Stoeny, 2019-01-27:
If I go into the Font Manager and go back, I can configure the OSD as before.
Is it a Font Issue?

DzikuVx, 2019-01-27:
This is a known bug that will be fixed. For now, for firmware family `2.0.x` use Configurator `2.0.x`.
Configurator `2.1.1` works only with `2.1` firmware

Kennymartin, 2019-01-27:
Thank you for the timely reply Pawel & I look forward to the fix.
Martin.

turbosi92, 2019-01-27:
I'm on the latest RC2 with 2.1.1 configurator on mac and the OSD will not load....FYI.  I read this as 2.1.1 works with latest 2.1 build but in my case it doesn't...at least on mac.

Older version of configurator on windows works fine.

jreise-d, 2019-01-27:
As posted above, start **configurator 2.1.0** and open OSD page once, that helps. After that configurator 2.1.1 and it's OSD page works fine with iNav 2.0.x and 2.1.0

lisdexia, 2019-01-28:
i am having the same issue with a mateksys f405 wing , the osd page is not loading inside the configurator however the osd is showing in the fpv feed, for the 2.0+ versions   however if i roll it all back to ver:1.9.1  it all works purfectly fine in the configurator and on the fpv osd feed.   
any news on a fix ?? id love to get flying.

axelradd, 2019-01-28:
Same issue on Matek F722 SE (Blank screen)

Frederx, 2019-01-29:
Having same issue with all F4 boards with 2.1.1 os Linux.

cmichael261, 2019-01-29:
I was originally running a Matek f722std on inav 2.0.0 with no problem with the OSD or font editor. Today I upgraded to Inav 2.1.1 and the OSD tab in the configurator is blank. I can see a OSD display in my goggles but I think it is a default, I can not change it. 
So I guess I'm having the same issue as everyone else.

mariniam, 2019-01-29:
same here with Matek f405 ctr. Empty OSD tab.

Nyanadayo, 2019-01-29:
Likewise with a Matek F405 CTR. I have reverted back to v2.0.0 Configurator and to 2.0.0 Firmware, now the OSD is back. I had the same issue with v2.2 and v2.1 configurator. Noticed also that current Chrome App on Ubuntu also has an empty OSD Tab..if you need more info let me know.

kd4kw, 2019-01-29:
What is strange is I tried to make some osd changes in cmd and it looked like they took but when I do a get osd its the same as before I tried to change them. I can rollback to any of the 1.9.x and the osd screen is fine with latest inav build

Kennymartin, 2019-02-02:
Is it possible to adjust the OSD configuration via cli commands?
Im having a problem re-installing an earlier version of inav configurator on my 64bit windows machine and therefore cannot scale back my setup to adjust OSD settings via the configurator tab. 
I want to enable the home arrow & GPS coordinates so that they are visable via my googles as current default settings do not include such data. 
Any help would be appreciated.

Kennymartin, 2019-02-02:
API version 2.20
FC Version 2.01

closed 2019-02-03 by DzikuVx

DzikuVx, 2019-02-03:
@Kennymartin you can build you own from the code or wait for next release which should happen relatively soon

Kennymartin, 2019-02-03:
No need to get bitchy now, you could of just said that in the first place.
Closing a thread issue in most peoples minds= fixed.

shellixyz, 2019-02-03:
@Kennymartin  He is not bitchy he told you what you needed to know to make it work. Or course it is not going to work in the present version of the configurator the fix have been merged only 2 hours ago. It is effectively fixed. Just not available yet in builds. So just be a little patient.

Kennymartin, 2019-02-03:
Thank you for the mannerly reply Michel.

Painless360, 2019-02-03:
Thanks for all the hard work guys on getting rid of these little annoying bugs. As a community we're lucky to have you doing all of this work for free.. Happy flying!

lisdexia, 2019-02-03:
YES AGREED , thankyou for your efforts. iNav is the best shit since cocaiahhhhhh sorry sliced bread

PR #671 (2019-02-03): fix OSD tab
Fixes #659