iNavFlight/inav-configurator #664: No map displayed on GPS tab with 2.1.1

Arvycka, 2019-01-26:
Map area in GPS tab remains blank at all times with the configurator 2.1.1:

[image]

Map is showing correctly  with configurator 2.0:

[image]

vasobg, 2019-01-29:
Yes, I have the same issue!

DzikuVx, 2019-02-03:
@Arvycka @vasobg 

Please post screenshots with settings popup:

[image]

"double gear" icon in top-right corner

Arvycka, 2019-02-03:
I don't have a GPS lock inside today, but even without it 2.0.0 displays a progress bar and a message "Waiting for GPS 3D lock".
There is nothing going on in that window in 2.1.1 - it's dead:

[image]

Arvycka, 2019-02-03:
Pawel, it's only in the Chrome App that the map is not displayed - it's fine in the stand-alone windows version.

closed 2019-02-11 by Arvycka