iNavFlight/inav-configurator #682: Generalize live update from FC to GUI

laurentopia, 2019-02-10:
## Current Behavior
Board sensor alignment roll pitch and yaw don't update as the FC is being adjusted by TX
As a result when pressing SAVE AND REBOOT in the configurator, all adjustment are reset to configurator settings

## Desired Behavior
All FC data live updates in the configurator, not only the 3D model of SETUP tab

## Suggested Solution
Add a read event in the adjustment fields

## Who does this impact? Who is this for?
Those who want to read their adjust rates while tuning in the TX

digitalentity, 2019-02-10:
Moved to Configurator project.

Configurator polls for config values when tab is loaded or when it expects values to change. This is by design, assumption is that if Configurator is connected, user will use the GUI to change values.

laurentopia, 2019-02-10:
I didn't see that option in the OSD, I'll use that.

To answer the question on why it's useful:
- consistent with setup tab which updates live, consistency is expected by the user
- helps set up the TX for adjust and in general know how much increase works and the direction

I realize it's not that trivial because you poll once and have a one-way input event -> send to FC.
Reusing what you have, this can be done like this:
1) display now look at a FC delta that gets polled as regularly as the heading in setup tab, if delta !=0 display the new value (even if user is editing)
2) if the polled delta = 0 and the user inputs a new value, send that new value and 1) will take care of displaying that value

MrD-RC, 2022-11-26:
Working as expected. Adjustments are for in-flight, not on the bench.

closed 2022-11-26 by MrD-RC