iNavFlight/inav-configurator #687: Configurator 2.1.2 Mixer tab displays bad values

noagendafpv, 2019-02-11:
Yesterday I used the iNav 2.1.2 configurator to setup my new F722-wing. I transfered my F405-wing smix via CLI, save/reboot, then went to the mixer tab and the values for the servos were all wrong, displaying the wrong servo numbers, wrong outputs, and weights that were well above 100. I then went back to CLI and did DIFF ALL and found that the smix numbers are correct for what I entered. The Mixer tab seems to have a bug.

shellixyz, 2019-02-11:
You need to update to 2.1.0-RC3 if you want to use version 2.1.2 of the configurator

closed 2019-02-11 by shellixyz

noagendafpv, 2019-02-11:
Desription "This version of INAV Configurator is intended to be used mainly with INAV firmware 2.1 family. Support for older versions of INAV might be limited." should reflect that. I was using a version of RC2