iNavFlight/inav-configurator #689: When a UART column is set, gray out the other ones

laurentopia, 2019-02-12:
## Current Behavior
When toggling UART2 RX to Serial RX I can still set Telemetry or Sensor or Peripherals, which shouldn't happen since a UART can only have one, but upon save and reboot that gets reset.

## Desired Behavior
gray out the column of the UART when one of them is alocated

## Suggested Solution
As soon as something in telemetry,rx, sensor or peripheral is set, gray out and disable input event from the other columns

## Who does this impact? Who is this for?
Everyone who starts in RC, and also I am sure sleep deprived veterans.
I am starting in RC and it took me 2 days of fumbling around before I stumbled on some archived post where one of the least upvoted answer explained as an afterthought that a UART can only do one RX or one TX at a time, not both. I was lucky to find that, and seeing the amount of posts trying to figure out telemetry on sbus, it's a common problem.

shellixyz, 2019-03-24:
It is not as easy because more than one protocol can be enabled on a port (port sharing) but I guess once serial RX is enabled on a port nothing else can so we could disable the other things in this case.

Tygrys-1, 2019-03-24:
> 
> It is not as easy because more than one protocol can be enabled on a port (port sharing) but I guess once serial RX is enabled on a port nothing else can so we could disable the other things in this case.

I have FlySky telemetry on the same port with the receiver.  And I do not know how it is with the FPort on FrSky. 
I think that the desired rule can be applied only to devices using Tx and Rx of the UART, like the i.e.  GPS.

Tygrys-1, 2019-03-24:
IMHO UARTS can do both input and output at the same time, but they share all the settings like the baud rate and parity ect. SPort telemetry uses different baud rate then the SBus, so this might be OP problem.

shellixyz, 2019-03-24:
> I have FlySky telemetry on the same port with the receiver

Ok then it is not possible.

> And I do not know how it is with the FPort on FrSky

IIRC you just need Serial RX enabled and FPort serial_rx protocol selected

> SPort telemetry uses different baud rate then the SBus

SBus and SPort can definitely not be used on the same UART

Tygrys-1, 2019-03-24:
> SBus and SPort can definitely not be used on the same UART

Yes, they cannot. 
[OT] There is a trick with the softserial and on Matek F405 there can be the SBus receiver on RX2 pad and SPort on TX2 pad. Resource-wise deign. And no inverter needed.

shellixyz, 2019-03-24:
Ok but then it is not technically the same UART

Tygrys-1, 2019-03-25:
> Ok but then it is not technically the same UART

Internally not. But from the outside it looks like ;-)

MrD-RC, 2022-11-26:
> > I have FlySky telemetry on the same port with the receiver
> 
> Ok then it is not possible.
> 

This closes the issue. Additional item needed on the same port as Serial RX with FlySky.

closed 2022-11-26 by MrD-RC