iNavFlight/inav-configurator #691: error message "need to upgrade..." after "save and reboot"

jreise-d, 2019-02-13:
In configurator 2.1.2 with activated Auto-connect after clicking "save and reboot" in any tab the fc is rebooting and reconnecting, but then disconnecting automatically with an error message in the configurator log "You need to upgrade your firmware before you can use the [...] tab.". This happens with my f411-wing with inav 2.1.0 RC3, but also with 2.0.1 stable.

Here is the log:
2019-02-13 @ 17:07:01 -- Running - OS: Windows, Chrome: 72.0.3626.96, Configurator: 2.1.2
2019-02-13 @ 17:07:04 -- Loaded release information from GitHub.
2019-02-13 @ 17:07:10 -- Serial port successfully opened with ID: 1
2019-02-13 @ 17:07:12 -- MultiWii API version received - 2.3.0
2019-02-13 @ 17:07:12 -- Flight controller info, identifier: INAV, version: 2.1.0
2019-02-13 @ 17:07:12 -- Running firmware released on: Feb 10 2019 16:46:25
2019-02-13 @ 17:07:12 -- Board: MK41, version: 0
2019-02-13 @ 17:07:12 -- Unique device ID received - 0x6500413437510c33353431
2019-02-13 @ 17:07:26 -- EEPROM saved
2019-02-13 @ 17:07:26 -- Device - Rebooting
2019-02-13 @ 17:07:27 -- Serial port successfully closed
2019-02-13 @ 17:07:28 -- Serial port successfully opened with ID: 2
2019-02-13 @ 17:07:30 -- MultiWii API version received - 2.3.0
2019-02-13 @ 17:07:30 -- Flight controller info, identifier: INAV, version: 2.1.0
2019-02-13 @ 17:07:31 -- Running firmware released on: Feb 10 2019 16:46:25
2019-02-13 @ 17:07:31 -- Board: MK41, version: 0
2019-02-13 @ 17:07:31 -- Unique device ID received - 0x6500413437510c33353431
2019-02-13 @ 17:07:33 -- Serial port successfully closed
2019-02-13 @ 17:07:34 -- **You need to upgrade your firmware before you can use the Configuration 
tab.**

Teraformed, 2019-02-15:
had this issue too

jreise-d, 2019-02-26:
It also happens with configurator 2.1.4 and inav 2.1.0 with activated Auto-connect. Tested on different computers. With **deactivated** Auto-connect it reconnects automatically after Save and Reboot and the error message didn't happen. Strange !?!
The Auto-connect function seems to be reversed since configurator 2.12

shellixyz, 2019-02-26:
What do you mean by reversed ?

DzikuVx, 2019-02-27:
You are mixing Auto-Connect with Reconnnect. Those are completely different things. Reconnect after save happens automatically and always.
Auto-Connect is designed to connect to a COM port that just became active, not reconnect.

DzikuVx, 2019-02-27:
For time being do not use Auto-Connect and that's all

jreise-d, 2019-02-28:
> 
> 
> You are mixing Auto-Connect with Reconnnect. Those are completely different things. Reconnect after save happens automatically and always.

In older versions before 2.1.2 reconnect after save and reboot didn't happen. I had to activate Auto-connect to achieve this.
Since version 2.1.2 I have to deactivate Auto-Connect to avoid the error message and achieve a Reconnect after save and reboot.
Thats what I mean by "reversed" or inverted

closed 2019-03-24 by shellixyz