iNavFlight/inav-configurator #700: 2.1 OSD temperature readout problem

japella, 2019-02-22:
After loaded 2.1 RC3 today. I selected the temperature elements for the OSD.

My problem is with the IMU and BARO temperature layout.
In the previous release candidates. It did not have a description next to the temperature readout.
Now that it does. There is a gap between the description and the temperature value.

The spacing is so wide that it seems like they are totally separate identity. 

Is this gap supposed to be there?  Can it be brought closer together?

Cheers

shellixyz, 2019-02-22:
The gap is supposed to be there. The labels are like that so that they are all aligned when you're enabling more than 1 temperature sensor. The gap is just enough so that it is possible to display the maximum temperature value (125°C) with the maximum label length of 4 chars. I'm going replace the labels with glyphs before 2.1 final is released it will take less space.

kardon18, 2019-02-22:
Okay. I understand why that is now. It just looks weird. 
The glyph sounds like the way to go with less space taken up. 
Thanks for the explanation.

PR #704 (2019-02-25): OSD: Add symbols for temperature sensor values
Configurator part for https://github.com/iNavFlight/inav/pull/4407

Closes #700
closed 2019-02-25 by shellixyz