iNavFlight/inav-configurator #702: pressing + on the GPS map pops back to default height

laurentopia, 2019-02-23:
instead of zooming in
repro:
click on gps tab or mission control
with openstreet on
click +
v2.1.3

laurentopia, 2019-03-03:
Still the case in 2.1.4
It prevents accurate mission setup, which becomes a bit more apparent when switching to Bing map with your own key.

Additional details on repros:
click on gps tab or mission control
with openstreet or bing ON
click +
the view will zoom in but then immediately snap back to the default zoomed out view
Also affected, zoom out [-] as well as drag map.
Essentially it is not possible to navigate the map
this happens on the v2.1.3 stand alone windows configurator

BenWThomas, 2019-04-02:
I have been experiencing this, too; is there a variable that can be set to control this?

DrJ-FPV, 2019-04-06:
I've had this as well, solution is available if the issue in this thread hasn't yet been resolved.

After different issues popped up, I changed iNav DPI behavior and this also resolved the zoom issue in Mission Control. This was Issue #741(closed). GPS Tab however still jumps as described. 

https://github.com/iNavFlight/inav-configurator/issues/741

Can we get a new feature? How about integrating 'show current location' on Mission Control Tab and rethinking the GPS tab. Addressing the new feature 'show current location' on the zoomable Mission Control Tab snuffs out the need for switching to GPS Tab in order to determine location during mission planning.

The feature would make the interface more user friendly AND resolve the jumping issue in the GPS Tab. Where's/who's the GUI guru?

[image]

BenWThomas, 2019-06-09:
After upgrading to iNav 2.2, GPS map pop-back to default height is still present.  (Tried Bing maps, but unsurprised that's affected, too.)  

Mission control also has a problem with the map - I get a blank blue field, and no map at all

[image]

DzikuVx, 2019-06-26:
fixed in#798 and will be available in the next Configurator release

closed 2019-06-26 by DzikuVx

BenWThomas, 2019-06-26:
Thanks for this and 702 — and thanks VERY much for the notification!

Regards

Ben W Thomas

J-Iszatt, 2020-07-15:
Hallo Team,

I am seeing the exact same bug on MacOs 10.14.6 in the Mission control with the 2.5.0 configurator.
Would be great if you could look at it again. I am planning to test older versions (2.3.0 in particular) to see if they work 
-> If it helps I can give feedback

Regards
JI

Marcus-Barcus, 2021-05-02:
Ive got a Bing Maps key but cant paste into area 
Also how do you set up Map Proxy & URL Layer 
Thanks