iNavFlight/inav-configurator #706: Matek 411 wing and INAV 2.1.0 Mixer problems

Kaffiflight, 2019-02-25:
Since three days I tried to make a wing ready with a Matek 411 wing and INAV 2.1.0. After many failures I found out the servomixer is not loading correctly. I can reproduce that issue, and I crosschecked it with 2.0.1, this does not happen there.
As follwos: Flashed, all settings to default (with "defaults" in CLI). Reboot. Go to mixertab, set "Airplane - Flying wing", load mixer. Save and reboot.
Check Mixer: It shows nonsense. Check CLI: It shows several rows of mixersettings, where only 4 lines should be.
As stated above, it works with INAV 2.0.1

DzikuVx, 2019-02-27:
@Kaffiflight CLI dump, screeshots, exact version of Configurator and firmware please

shellixyz, 2019-02-27:
@Kaffiflight I guess you are using an older version of the configurator with iNav 2.1.0. You need to use at least version 2.1.4 of the configurator with 2.1.0.

Kaffiflight, 2019-02-28:
Thank you for your interest and feedback. I can tell I found the reason why.
Yes, I used the configurator 2.1.3, but then just when it came out, the 2.1.4.
The Problem was elsewhere: 
I flashed with a hexfile from my hardisk that I downloaded before (2.1.0 RC3 for the Matek 411, standard file). It flashed without showing any Problems.
Yesterday I downloaded the same file again, overwriting the old one. Now it works like it should be. Obviously there was a faulty hexfile on my PC - must have happened during download. I flashed with it maybe 10 times on 2 different PCs, always with the same result. Not showing that anything was wrong. 
Thanks.

closed 2019-02-28 by Kaffiflight

shellixyz, 2019-02-28:
The problem might have been that you used a version of the FW older than 2.1.0-RC3 with configurator version above 2.1.3 that doesn't work either