iNavFlight/inav-configurator #709: [REQUEST] invert yaw,pitch & roll on a switch

Teraformed, 2019-03-02:
I don't know if this is already possibe.

Considering more and more F7 are coming out with 2 cams input where you can toggle between both with a switch. We'll will see more and more quads consisting of front and back camera.

I just build a quad with extra camera facing backwards. Obviously the stick input won't behave the same way like when you are viewing the camera facing front.

I think this feature would be nice to have.

It should be possible to invert yaw, pitch & roll all at same time when toggling to the second camera facing backwards.

fahrvergnuugen, 2019-03-04:
Couldn't you do this in the radio?

shellixyz, 2019-03-04:
This request belongs to the firmware not the configurator

Teraformed, 2019-03-06:
> Couldn't you do this in the radio?

Well kinda but not on a switch with my current radio.

> This request belongs to the firmware not the configurator

I thought because setting up "Modes" in configurator this belonged here.

DzikuVx, 2019-03-07:
This is firmware and VTOL tailsitter this way or another is coming to INAV.

closed 2019-03-07 by DzikuVx