iNavFlight/inav-configurator #712: Modeled a Wing to replace the default Box

laurentopia, 2019-03-07:
Attached is the obj, you can chose the format in this page https://a360.co/2TiEye2
[image]

[S800 wing.zip](https://github.com/iNavFlight/inav-configurator/files/2941048/S800.wing.zip)

Arvycka, 2019-03-07:
A fixed wing model of any sort would be better than that ancient ugly box :)

laurentopia, 2019-03-09:
> 
> 
> A fixed wing model of any sort would be better than that ancient ugly box :)

Except that

laurentopia, 2019-03-09:
> 
> 
> A fixed wing model of any sort would be better than that ancient ugly box :)

The Fusion360 repository is public, feel free to change the model.

Arvycka, 2019-03-10:
I meant the box in the configurator, your model is fine ☺

Mateyhv, 2020-02-05:
Any way to replace the box with the S800 wing obj?

DzikuVx, 2020-02-06:
It's a beuty of a open source that anybody can do it

closed 2021-08-13 by DzikuVx