iNavFlight/inav-configurator #717: .desktop

Zipdox, 2019-03-10:


Zipdox, 2019-03-10:
I wrote a little desktop file that I think should be added to the Linux versions.

```
[Desktop Entry]
Name=Inav Configurator
Exec=bash -c 'cd "$(dirname "$0")"; ./inav-configurator' %k
Type=Application
```

shellixyz, 2019-03-10:
I don't think you need to cd. But maybe there a good reason you it for ? Good idea to add a desktop file

Zipdox, 2019-03-10:
> I don't think you need to cd. But maybe there a good reason you it for ? Good idea to add a desktop file
.desktop file paths are absolute so this trick is mandatory or it won't work

wx4cb, 2020-04-13:
you don't have to put a "cd" in, just run it with the absolute path. in my case

```
[Desktop Entry]
Name=Inav Configurator
Exec=bash -c '/usr/src/RC/INAVConfigurator/inav-configurator' %k
Type=Application
```